mikakaraila/node-red-contrib-opcua #84: OPC UA connection error: Good (0x00000)

vgezer, 2018-06-15:
I am using Node red 0.18.7 on Debian with this palette v0.2.31.

Before updating it into the latest version, I had no problem with connection except for some unexpected disconnects. Today, I updated it via node-red, and now all opcua-client nodes in flows show error. Here is the console message:

~~~

15 Jun 12:02:38 - [error] [OpcUa-Client:XXX] XXX OPC UA connection error: Good (0x00000)
15 Jun 12:02:38 - [error] [OpcUa-Client:YYY] YYY OPC UA connection error: Good (0x00000)
15 Jun 12:02:38 - [error] [OpcUa-Client:CCC] CCC OPC UA connection error: Good (0x00000)
15 Jun 12:02:38 - [error] [OpcUa-Client:DDD] DDD OPC UA connection error: Good (0x00000)
15 Jun 12:02:38 - [error] [OpcUa-Client:EEE] EEE OPC UA connection error: Good (0x00000)
~~~

I am able to use another opc-ua client and read out the values.
I also tried to use subscribe, but it is the same. As soon as I deploy the flow, these are what I only see.

How can I investigate further?

mikakaraila, 2018-06-18:
Can you start flow with verbose option?
node-red -v TEST-FLOW.json

vgezer, 2018-06-18:
Here is the log:

~~~
18 Jun 10:12:06 - [info] Starting flows
18 Jun 10:12:06 - [info] Started flows
18 Jun 10:12:06 - [info] Server now running at http://127.0.0.1:1881/
18 Jun 10:13:39 - [info] Stopping flows
18 Jun 10:13:39 - [info] Stopped flows
18 Jun 10:13:39 - [info] Starting flows
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] [object Object]
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] [object Object]
18 Jun 10:13:39 - [warn] [OpcUa-Client:866b4017.ab1f6] OpcUaClientNode: create C                                                        lient ...
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] [object Object]
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] Client status: create cli                                                        ent
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] async series - connecting                                                        
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] Client status: connecting
18 Jun 10:13:39 - [info] Started flows
18 Jun 10:13:39 - [error] [OpcUa-Client:866b4017.ab1f6]  OPC UA connection error                                                        : Good (0x00000)
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] Error: Good (0x00000)
    at /home/mitarbeiter/.node-red/node_modules/node-opcua/node_modules/node-opc                                                        ua-secure-channel/src/client/client_secure_channel_layer.js:508:21
    at modified_callback (/home/mitarbeiter/.node-red/node_modules/node-opcua/no                                                        de_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1                                                        005:28)
    at process_request_callback (/home/mitarbeiter/.node-red/node_modules/node-o                                                        pcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_lay                                                        er.js:78:5)
    at ClientSecureChannelLayer._on_message_received (/home/mitarbeiter/.node-re                                                        d/node_modules/node-opcua/node_modules/node-opcua-secure-channel/src/client/clie                                                        nt_secure_channel_layer.js:129:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (/home/mitarbeiter/.node-red/node_mod                                                        ules/node-opcua/node_modules/node-opcua-secure-channel/src/message_builder.js:46                                                        1:22)
    at MessageBuilderBase._feed_messageChunk (/home/mitarbeiter/.node-red/node_m                                                        odules/node-opcua/node_modules/node-opcua-transport/src/message_builder_base.js:                                                        196:18)
    at PacketAssembler.<anonymous> (/home/mitarbeiter/.node-red/node_modules/nod                                                        e-opcua/node_modules/node-opcua-transport/src/message_builder_base.js:53:14)
    at emitOne (events.js:116:13)
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] Client status: connection                                                         error
18 Jun 10:13:39 - [info] [OpcUa-Client:866b4017.ab1f6] Client disconnected!
~~~

mikakaraila, 2018-06-18:
Hmm, no idea yet. Are you using some securityMode / securityPolicy?

vgezer, 2018-06-19:
No. Is there a way to downgrade it into 0.2.22? I tried to use npm node-xxx@0.2.22, but apparently it did not downgrade. Might it be because it is in place?

mikakaraila, 2018-06-19:
Try older, I skipped be accident one version.

vgezer, 2018-06-27:
I think I have some progress. I downgraded Node-Red into 0.17.5 and opcua node to 0.2.20. However I don't know why I receive the following exception. Do you know why I need a certificate for non-secure connection? And why it shows this exception now, although I had no certificate before as well?

~~~
Welcome to Node-RED
===================

27 Jun 16:36:40 - [info] Node-RED version: v0.17.5
27 Jun 16:36:40 - [info] Node.js  version: v8.11.3
27 Jun 16:36:40 - [info] Linux 3.16.0-4-amd64 x64 LE
27 Jun 16:36:40 - [info] Loading palette nodes
27 Jun 16:36:42 - [warn] ------------------------------------------------------
27 Jun 16:36:42 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
27 Jun 16:36:42 - [warn] ------------------------------------------------------
27 Jun 16:36:42 - [warn] Missing node modules:
27 Jun 16:36:42 - [warn]  - node-red-contrib-iiot-opcua (2.7.1): OPCUA-IIoT-Server-ISA95
27 Jun 16:36:42 - [info] Removing modules from config
27 Jun 16:36:42 - [info] Settings file  : /home/mitarbeiter/.node-red/settings.js
27 Jun 16:36:42 - [info] User directory : /home/mitarbeiter/.node-red
27 Jun 16:36:42 - [info] Flows file     : /home/mitarbeiter/.node-red/TEST-FLOW.json
27 Jun 16:36:42 - [info] Server now running at http://127.0.0.1:1881/
27 Jun 16:36:42 - [info] Starting flows
27 Jun 16:36:42 - [info] [OpcUa-Client:xxx] [object Object]
27 Jun 16:36:42 - [warn] [OpcUa-Client:xxx] xxx: create Client ...
27 Jun 16:36:42 - [info] [OpcUa-Client:xxx] [object Object]
27 Jun 16:36:42 - [error] [OpcUa-Client:d03d8253.8baa7] Error:  cannot locate certificate file /usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/certificates/client_selfsigned_cert_1024.pem
    at OPCUAClientBase (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/client_base.js:68:15)
    at new OPCUAClient (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/opcua_client.js:74:21)
    at create_opcua_client (/usr/lib/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:148:21)
    at new OpcUaClientNode (/usr/lib/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:275:5)
    at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:305:18)
    at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:89:35)
    at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:300:29)
    at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)
    at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)
    at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)

~~~

vgezer, 2018-06-27:
Used https://github.com/node-opcua/node-opcua/issues/254#issuecomment-255598210 and fixed the error and now old version works.

mikakaraila, 2018-06-29:
> Used node-opcua/node-opcua#254 (comment)
> and fixed the error and now old version works.

Yes, installing npm -g causes this error. There is some other bugs in
node-opcua v0.4.1 and I´m waiting new version.