mikakaraila/node-red-contrib-opcua #86: Browsing Issues

iba-dchawla, 2018-08-16:
Hello,

I am using the following flow to browse tags using OPC UA. I was successfully able to read tags, but I don't see any errors or messages when I try to browse. I am new to both Node Red and OPC UA. Any help would be appreciated. 

```
[
    {
        "id": "b4998c7d.25656",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "db20968b.4637c8",
        "type": "inject",
        "z": "b4998c7d.25656",
        "name": "Rolling Load",
        "topic": "ns=3;s=V:0.3.10.0.1;datatype=Float",
        "payload": "",
        "payloadType": "date",
        "repeat": "5",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 221,
        "y": 95,
        "wires": [
            [
                "41e097a.f939368"
            ]
        ]
    },
    {
        "id": "41e097a.f939368",
        "type": "OpcUa-Client",
        "z": "b4998c7d.25656",
        "endpoint": "132f6195.f7ddbe",
        "action": "read",
        "time": 10,
        "timeUnit": "s",
        "name": "Diksha's PDA OPC Server",
        "x": 516,
        "y": 80,
        "wires": [
            [
                "51fb83ad.9ad5ac"
            ]
        ]
    },
    {
        "id": "51fb83ad.9ad5ac",
        "type": "debug",
        "z": "b4998c7d.25656",
        "name": "Rolling Load",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "x": 775,
        "y": 81,
        "wires": []
    },
    {
        "id": "dc3d7b50.219d88",
        "type": "OpcUa-Client",
        "z": "b4998c7d.25656",
        "endpoint": "132f6195.f7ddbe",
        "action": "browse",
        "time": 10,
        "timeUnit": "s",
        "name": "",
        "x": 525.433349609375,
        "y": 244,
        "wires": [
            [
                "5a5cbfff.83f2a"
            ]
        ]
    },
    {
        "id": "b8acf301.4fac9",
        "type": "inject",
        "z": "b4998c7d.25656",
        "name": "Browse",
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": "",
        "x": 144,
        "y": 244.86664581298828,
        "wires": [
            [
                "68661315.c506ec"
            ]
        ]
    },
    {
        "id": "68661315.c506ec",
        "type": "function",
        "z": "b4998c7d.25656",
        "name": "Set browse address",
        "func": "msg.topic=\"ns=3;s=V\";\nmsg.actiontype=\"browse\";\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 314.00001525878906,
        "y": 243.066650390625,
        "wires": [
            [
                "dc3d7b50.219d88"
            ]
        ]
    },
    {
        "id": "fc9ce3d5.feed8",
        "type": "OpcUa-Client",
        "z": "b4998c7d.25656",
        "endpoint": "132f6195.f7ddbe",
        "action": "subscribe",
        "time": "10",
        "timeUnit": "s",
        "name": "",
        "x": 905.0001754760742,
        "y": 245.86665153503418,
        "wires": [
            [
                "5d31066c.d44ca8"
            ]
        ]
    },
    {
        "id": "5d31066c.d44ca8",
        "type": "debug",
        "z": "b4998c7d.25656",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "x": 1061.000072479248,
        "y": 245.8666534423828,
        "wires": []
    },
    {
        "id": "5a5cbfff.83f2a",
        "type": "function",
        "z": "b4998c7d.25656",
        "name": "nodeId -> Topic",
        "func": "if (msg.nodeId.toString().indexOf(\"ns=3\")>=0) {\n  msg.topic=msg.nodeId;\n    return msg;\n}",
        "outputs": 1,
        "noerr": 0,
        "x": 707.2333450317383,
        "y": 245.0000343322754,
        "wires": [
            [
                "fc9ce3d5.feed8"
            ]
        ]
    },
    {
        "id": "132f6195.f7ddbe",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://iba-607991:48080",
        "secpol": "Basic256",
        "secmode": "SIGNANDENCRYPT",
        "login": true
    }
]
```

mikakaraila, 2018-08-20:
Seems to work, but what you want to get? I can study more...

dchawla7, 2018-08-20:
I would like to be able to read tag names (Cosine, Sine, Digital Signal 1, Digital Signal 2, Saw Tooth) as shown in the screenshot below:

[image]

When I specify node name ns=3;s=F:0.3 (which as you can see from the screenshot above is the node name of the Modules folder which has the Virtual 1 and Virtual 2 modules) to OPC UA browser, it returns 4 objects/items - Root, Server, PlaceHolder and FolderType. 

[{"item":{"referenceTypeId":"ns=0;i=35","isForward":false,"nodeId":"ns=0;i=84","browseName":{"namespaceIndex":0,"name":"Root"},"displayName":{"text":"Root"},"nodeClass":"Object","typeDefinition":"ns=0;i=61"}},{"item":{"referenceTypeId":"ns=0;i=35","isForward":true,"nodeId":"ns=0;i=2253","browseName":{"namespaceIndex":0,"name":"Server"},"displayName":{"text":"Server"},"nodeClass":"Object","typeDefinition":"ns=0;i=2004"}},{"item":{"referenceTypeId":"ns=0;i=35","isForward":true,"nodeId":"ns=2;s=F:0","browseName":{"namespaceIndex":2,"name":"PlaceHolder" (F:0)"},"displayName":{"text":"ibaPDA","locale":""},"nodeClass":"Object","typeDefinition":"ns=0;i=61"}},{"item":{"referenceTypeId":"ns=0;i=40","isForward":true,"nodeId":"ns=0;i=61","browseName":{"namespaceIndex":0,"name":"FolderType"},"displayName":{"text":"FolderType"},"nodeClass":"ObjectType","typeDefinition":"ns=0;i=0"}}]

When I inject again it returns 4 more items - 5 Virtual Mode 1, 8 Virtual Module 2, FolderType, PlaceHolder

[{"item":{"referenceTypeId":"ns=0;i=35","isForward":true,"nodeId":"ns=3;s=F:0.3.5","browseName":{"namespaceIndex":3,"name":"5 Virtual Module 1 (F:0.3.5)"},"displayName":{"text":"5 Virtual Module 1"},"nodeClass":"Object","typeDefinition":"ns=0;i=61"}},{"item":{"referenceTypeId":"ns=0;i=35","isForward":true,"nodeId":"ns=3;s=F:0.3.8","browseName":{"namespaceIndex":3,"name":"8 Virtual Module 2 (F:0.3.8)"},"displayName":{"text":"8 Virtual Module 2"},"nodeClass":"Object","typeDefinition":"ns=0;i=61"}},{"item":{"referenceTypeId":"ns=0;i=40","isForward":true,"nodeId":"ns=0;i=61","browseName":{"namespaceIndex":0,"name":"FolderType"},"displayName":{"text":"FolderType"},"nodeClass":"ObjectType","typeDefinition":"ns=0;i=0"}},{"item":{"referenceTypeId":"ns=0;i=35","isForward":false,"nodeId":"ns=2;s=F:0","browseName":{"namespaceIndex":2,"name":"PlaceHolder (F:0)"},"displayName":{"text":"ibaPDA","locale":""},"nodeClass":"Object","typeDefinition":"ns=0;i=61"}}]

What node name do I need to specify for it to return tag names (Cosine, Sine, Digital Signal 1, SawTooth etc.) instead of the folder names.

Thanks again!

mikakaraila, 2018-08-20:
This example will browse address space and filter items & subscribe them.
[image]

dchawla7, 2018-08-20:
Thanks! Let me try that and get back with you.

dchawla7, 2018-08-20:
I tried to import the flow, but the import button is grayed out.

mikakaraila, 2018-08-21:
Did you copy snippet into clipboard?

dchawla7, 2018-08-21:
Yes, I did. 

I looked at some of your old posts ( see link below) and you said copying the code seems to add new lines. I did remove the new lines, but still had issues. It would be awesome if you could send it to me in a file.

https://github.com/mikakaraila/node-red-contrib-opcua/issues/31

Thanks!

mikakaraila, 2018-08-22:
Here (rename file .txt -> .json). Tip look function node after browse.
[OPCUA_BROWSE_and_SUBSCRIBE3.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/2309018/OPCUA_BROWSE_and_SUBSCRIBE3.txt)

greatdsa, 2019-04-11:
@mikakaraila I have used Prosys Simulation Server and followed the same flow:
https://imgur.com/a/NcCF8h3

But I am not getting any item on browsing.
Thanks

mikakaraila, 2020-07-07:
New version makes better browse

closed 2020-07-07 by mikakaraila