mikakaraila/node-red-contrib-opcua #87: Invalid namespace

cabsil, 2018-08-23:
Hello,

I use the node-red-contrib-opcua in my company. This is a great tool saving a lot of work !! ;-)

But when I start node-red with one opcua server node, node-red crashes with a message with "invalid namespace". After some invetigation, we commented the call to construct_my_address_space in 104-opcuaserver and it solves the issue. This method looks like a kind of demo structure.

Is this normal ? Do we have done something wrong ? 

Regards

Christophe Absil

mikakaraila, 2018-08-24:
I am using this kind of flow and I can add objects at runtime.
[image]

mikakaraila, 2018-08-24:
Here is the flow file (rename .txt -> .json).
[OPCUA-SERVER.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/2317341/OPCUA-SERVER.txt)

cabsil, 2018-08-24:
Hello,

How to reproduce : on a fresh install of node-red and node opcua. I have a flow with one single opcua server. When I deploy this flow, nodered crashes with the error. Then I am unable to start it over. 

I have to say that I have another PC where I have exactly the same installation and I didn't have the problem. 

Regards

Christophe Absil

mikakaraila, 2018-08-24:
Hmm, then it seems to be somehow environment problem. 
Can you run "node-red -v OPCUA_SERVER.json" to see some output?
Default address space:
[image]

kurceliana, 2018-08-24:
Hello,

I have the same problem (the server-node won't start) when using a single server-node or the example file due to an invalid namespace error.

I am using Node-RED on a Raspberry Pi (newest Raspbian + fresh install of Node-RED and node-red-contrib-opcua). 

Maybe the error message from the Node-RED console will help:

Warning !  AddressSpace#addObject  is now deprecated
          use addressSpace.getOwnNamespace().addObject(..) instead
-----------------------------------------------------------
Error: invalid namespace index ( out of bound)
    at assert (/home/pi/.node-red/node_modules/node-opcua-assert/dist/C:/sterfive/NodeOPCUA/test-node-opcua/node-opcua/packages/node-opcua-assert/source/index.ts:5:21)
    at AddressSpace.getNamespace (/home/pi/.node-red/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/home/pi/.node-red/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addObject (/home/pi/.node-red/node_modules/node-opcua-address-space/src/address_space.js:969:17)
    at AddressSpace.Constructor.(anonymous function) [as addObject] (/home/pi/.node-red/node_modules/node-opcua-utils/dist/C:/sterfive/NodeOPCUA/toto/node-opcua/packages/node-opcua-utils/source/set_deprecated.ts:16:27)
    at construct_my_address_space (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:39)
    at post_initialize (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Unit entered failed state.
nodered.service: Failed with result 'exit-code'.
nodered.service: Service hold-off time over, scheduling restart.
Stopped Node-RED graphical event wiring tool.

Reboot and new installations haven't fixed it yet.

Regards

Kurc

Estuper, 2018-08-25:
Hello, 

I've experienced the same error with a clean installation of Ubuntu 16.04 over Odroid XU4. I get the following error:

Warning !  AddressSpace#addObject  is now deprecated
          use addressSpace.getOwnNamespace().addObject(..) instead
-----------------------------------------------------------
Error: invalid namespace index ( out of bound)
    at assert (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addObject (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:969:17)
    at AddressSpace.Constructor.(anonymous function) [as addObject] (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:39)
    at post_initialize (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
    at Immediate.<anonymous> (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
-----------------------------------------------------------
25 Aug 15:37:39 - [red] Uncaught Exception:
25 Aug 15:37:39 - Error: invalid namespace index ( out of bound)
    at assert (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addObject (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:969:17)
    at AddressSpace.Constructor.(anonymous function) [as addObject] (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:39)
    at post_initialize (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
    at Immediate.<anonymous> (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)

After that I have tried to use the recomended command "node-red -v OPCUA_SERVER.json" and I get the same output error:

Warning !  AddressSpace#addObject  is now deprecated
          use addressSpace.getOwnNamespace().addObject(..) instead
-----------------------------------------------------------
Error: invalid namespace index ( out of bound)
    at assert (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addObject (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:969:17)
    at AddressSpace.Constructor.(anonymous function) [as addObject] (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:39)
    at post_initialize (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
    at Immediate.<anonymous> (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
-----------------------------------------------------------
25 Aug 15:43:38 - [red] Uncaught Exception:
25 Aug 15:43:38 - Error: invalid namespace index ( out of bound)
    at assert (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at AddressSpace.getNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:116:9)
    at AddressSpace._resolveRequestedNamespace (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:964:17)
    at AddressSpace.addObject (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-address-space/src/address_space.js:969:17)
    at AddressSpace.Constructor.(anonymous function) [as addObject] (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-utils/dist/set_deprecated.js:15:27)
    at construct_my_address_space (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:39)
    at post_initialize (/usr/lib/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
    at Immediate.<anonymous> (/usr/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua-server/src/opcua_server.js:625:9)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)

I hope this information will be useful.

Best regards,

Francisco,

kurceliana, 2018-08-25:
Hello again,

I tried a little and substituted all addressSpace.addSomethings() on my system with addressSpace.getOwnNamespace().addSomethings() in the 104-opcuaserver.js file, which led to a new error message:

24 Aug 15:57:42 - Error: node must belongs to this namespace
    at UANamespace._register (/home/pi/.node-red/node_modules/node-opcua-address-space/src/namespace.js:217:15)
    at UANamespace._createNode (/home/pi/.node-red/node_modules/node-opcua-address-space/src/namespace.js:415:10)
    at /home/pi/.node-red/node_modules/node-opcua-address-space/src/namespace.js:942:21
    at AddressSpace.modelChangeTransaction (/home/pi/.node-red/node_modules/node-opcua-address-space/src/address_space.js:933:9)
    at UANamespace.createNode (/home/pi/.node-red/node_modules/node-opcua-address-space/src/namespace.js:924:18)
    at UANamespace.addObject (/home/pi/.node-red/node_modules/node-opcua-address-space/src/namespace.js:665:22)
    at construct_my_address_space (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:108:57)
    at post_initialize (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/104-opcuaserver.js:232:17)
    at Immediate._onImmediate (/home/pi/.node-red/node_modules/node-opcua-server/src/opcua_server.js:625:9)

Since I am lucky enough to have a slightly older Pi, where the OPC UA Server starts fine, I checked which of the mentioned files may have changed. The 104 file hasn't changed and the opcua_server.js file (see error message) has undergone some exception catching, whereas the namespace.js file from the node-opcua-address-space module has been updated significantly.

Maybe these changes triggered the bug (I'll probably try further on monday).

I hope this may help.

Regards

Kurc

mikakaraila, 2018-08-26:
I suppose you have upgraded node-opcua to v4.2 and it will cause this. Next v5.0 node-opcua will use ownAddressSpace. My example puts some variables to ns=4 and it causes this.
So you can edit code to use ns=1 and it should fix this. I will take a look later.

mikakaraila, 2018-08-27:
I found another place at namespace.js line 389:
 assert(options.browseName.indexOf(":") === -1, "We do not support <namespace>:<browseName> form here yet");
This is causing the current problems.

mikakaraila, 2018-08-27:
node-opcua v4.3 will fix assert, but I will add getOwnNamespace() and fix browserNames.

Estuper, 2018-08-27:
Hi again, I have done every change you have purposed and I have not failures any more.
[image]
 However, I cant either search or read the server from UAexpert. Anybody has experienced the same problem?
Many thanks in advance

mikakaraila, 2018-08-27:
Yes it is already reported. V4.3 is still broken...

mikakaraila, 2018-08-28:
Fix is ready in github, please test. I will publish to npm after somebody has tested.

Estuper, 2018-08-30:
Hi again,

I've been testing it and when the system is working for a long I get the same issue:
[image]
[image]

I have no idea wheter the update is provocating this failure.

Many thanks in advance 

I am going to keep up testing it.

mikakaraila, 2018-09-03:
For me your last error is totally something else than invalid namespace index. It is not node-red related instead somethig with node-opcua readRequest message encoding. Which server you are using?

mikakaraila, 2018-09-04:
Original error fixed -> this issue solved. Open new case if some other error occurs.

closed 2018-09-04 by mikakaraila

Jelazzaoui, 2020-02-11:
Hallo maikakraila, I am using node-red-contrib-opcua client to write and read Data from PLC S7-1512. Reading data works with no problems but when i try to write in the PLC i receive this errors:
`Error
    at assert (/home/pi/.node-red/node_modules/node-opcua-assert/dist/index.js:6:21)
    at new NodeId (/home/pi/.node-red/node_modules/node-opcua-nodeid/src/nodeid.js:86:5)
    at write_action_input (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:722:18)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:429:11)
    at OpcUaClientNode.Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:199:18)
    at Immediate._onImmediate (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:179:33)
    at processImmediate (internal/timers.js:439:21)`

I wish you could help me to solve this issue.
Best regards 
Jalal
[image]

yvanchkouropadsky, 2021-01-12:
Hello All and Happy new year!!
using this template, i can not understand how you can create new tags under a new equipment created??
for exemple, to create a "speed" tag with "30" in value, under the "machine1" device created with the inject button??

Thanks if you can hepl me!
best.

mikakaraila, 2021-01-12:
Here example:
[image]

addFolder:
[image]

addVariable:
[image]

yvanchkouropadsky, 2021-01-12:
Thanks a lot @mikakaraila !!
it works well!