mikakaraila/node-red-contrib-opcua #89: OPC UA Connection Timeout

mandeepahuja, 2018-09-07:
Hello, I am using the OPC UA node and it's working great. However the issue I am running into is that if there is ever a connection error, it never resets or re-tries or times out. If I simply redploy the flow it starts to work again. Is there something I am missing? Where can I specify a connection timeout for the OPCUa-Client Node? Please help!

mikakaraila, 2018-09-07:
In the code connectionStrategy has following options:

connectionOption.connectionStrategy= {
        maxRetry: 10512000,      // 10 years should be enough. No infinite parameter for backoff.
        initialDelay: 2000,
        maxDelay: 30*1000
    };

mandeepahuja, 2018-09-07:
Sorry can you point me where is this? Settings file? or something I have to write to the Client node?

mikakaraila, 2018-09-07:
It is in the 102-opcuaclient.js file.

mandeepahuja, 2018-09-07:
Ok what do you recommend I set the parameters as?

mikakaraila, 2018-09-08:
I still do not understand your problem. What is server? How to reproduce state?
You can try smaller values to see what happens...

mandeepahuja, 2018-09-13:
Hi Mika,
Please see the screenshot below. It gets into this mode and never recovers from it. However if I re-deploy it will immediately connect. Not sure how to explain why it is not re-trying for a connection after a failure.

Any chance you’ve seen this before or how to get around it?

[cid:image003.png@01D44B78.35C1D080]

[ifmLogo]

Mandeep Ahuja
Solutions Manager

+1 610 314 8130

mikakaraila, 2018-09-14:
Screenshot picture is not loading, can you update your previous comment?

mandeepahuja, 2018-09-14:
[image]

mikakaraila, 2018-10-15:
Have you run node-red in verbose mode to see more what is causing actual error?
node-red -v YOURFLOW.json

hj91, 2019-05-01:
> It is in the 102-opcuaclient.js file.
> 
> How to get into state that client does not re-connect?

To produce this issue, disconnect your opc server (in my case its kepware running on windows box) and after 5 minutes, if opc srver is not back online, the node-red flow crashes.

mikakaraila, 2020-07-07:
I expect this is not anymore active

closed 2020-07-07 by mikakaraila