mikakaraila/node-red-contrib-opcua #91: install error Unexpected end of JSON input while parsing near '...{"name":"loose-envify'

projjwal1975, 2018-11-04:
"npm install node-red-contrib-opcua" is throwing above error. I am using node v10.13.0 & npm 6.4.1. 

attached here's the screenshot
[image]

also the log file
[2018-11-04T19_45_49_449Z-debug.log](https://github.com/mikakaraila/node-red-contrib-opcua/files/2546239/2018-11-04T19_45_49_449Z-debug.log)

Kindly help

Regards

Lothdeorn, 2018-11-06:
It's maybe a stupid question do you have nodejs install ? 
Also, if you want to use it with node-red, you need to install this flow in the node-red folder 

C:\USERS\USER_NAME\.node-red\

projjwal1975, 2018-11-07:
thanks for responding. Unfortunately, I had tried your way & got same error. Node-red is running perfectly ok in my PC, but this opcua node. I am using node v10.13.0. I had previously installed this node successfully in my other PC that was running node v8.x.y. I updated this other PC to v10.13.0 & the same node is still working there. If you need any further info kindly ask. Regards

mikakaraila, 2019-03-06:
I suppose there has been some other package using loose-envify@^1.0.0 that had contained wrongly written JSON. Just try to reinstall, perhaps somebody has fixed it.

closed 2019-03-06 by mikakaraila