mikakaraila/node-red-contrib-opcua #92: Adding possibility to put IP address in input

Lothdeorn, 2018-11-06:
Hello,

I love this mode but i think it would be great to be able to have an input dedicated to ip adress. 
To be able to use the same block for multiple device.

I didn't find a way to have a string containing the ip address on Node-Red. 

Maybe there is a trick to do it in the endpoint field.

[image]

mikakaraila, 2018-11-06:
Good idea, I have to check if this will cause something inside code...

Lothdeorn, 2018-11-06:
I've success to did it with the trick proposed in a closed issue. 

https://github.com/mikakaraila/node-red-contrib-opcua/issues/59

mikakaraila, 2018-11-06:
Good, I cannot remember everything what I have implemented :-)
I have been implementing OPC UA Server & Client at work.

closed 2018-11-06 by mikakaraila