mikakaraila/node-red-contrib-opcua #95: Question : Write to DateTime field

craftery, 2018-12-11:
Hi,
Is it possible to write to a opc item configured as a DateTime? I can read from the item okay but when I try to use the same payload to write back to it I don't get any error but the write does not succeed.
This is the payload I get when I read the opcitem
{"_msgid":"60b6fe90.68cbe","topic":"ns=2;s=Channel1.Device1.Milling.dateofadditions","payload":"2017-12-30T00:00:00.000Z","datatype":"DateTime","browseName":"dateOfAdditions"}
And when I try to pass the following payload nothing happens
{"topic":"ns=2;s=Channel1.Device1.Milling.dateofadditions","payload":"2017-12-30T00:00:00.000Z","datatype":"DateTime","browseName":"dateOfAdditions"}
I also tried passing a javascript date (ticks since 1970) in place of the datetime UTC formatted string to no avail.
Any pointers would be appreciated.
Thanks,
Cormac

mikakaraila, 2018-12-11:
Hmm, it should work as in code it will be handled:
     case "DateTime":
            nValue = new opcua.Variant({ dataType: opcua.DataType.UtcTime, value: Date.parse(value) });
            break;
I will take a look...

mikakaraila, 2018-12-11:
In my test flow it works:
[image]

DateTime is inject in this format:
[image]

mikakaraila, 2018-12-11:
Sorry, did not actually write it to server, small fix needed:
		case "DateTime":
            nValue = { dataType: opcua.DataType.DateTime, value: new Date(value) };
            break;

mikakaraila, 2018-12-11:
I published new version 0.2.33

mikakaraila, 2018-12-11:
I tested with payload in ISOstring format: 2018-10-11T12:11:10.000Z

closed 2018-12-11 by mikakaraila