mikakaraila/node-red-contrib-opcua #96: Multiple IP injection issue

Lothdeorn, 2018-12-13:
Hello,

I've made a code to be able to inject IP in the OPC UA Client. On the paper, it seem working. I see than the debug msg show the request sended in good order (Like in issue#59), see the picture below. But i don't know why, on the output of OPCUA client, it send me always the value of the variable of the first IP. Even if the msg object is correctly sended. Do you know what is wrong in my code ? 

[image]

I've exported my code : 

`[{"id":"2927dc06.684b3c","type":"tab","label":"Flow 3","disabled":false,"info":""},{"id":"4c4177c8.4a4bd8","type":"inject","z":"2927dc06.684b3c","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":470,"y":240,"wires":[["ad225a5c.1f762"]]},{"id":"ad225a5c.1f762","type":"function","z":"2927dc06.684b3c","name":"IP CLIENT","func":"msg.topic = \"SELECT ip_client FROM `liste_machines`\" ;\n\nreturn msg;","outputs":1,"noerr":0,"x":650,"y":240,"wires":[["5398a0b8.935ea8"]]},{"id":"5398a0b8.935ea8","type":"mysql","z":"2927dc06.684b3c","mydb":"2319a0e1.5a665","name":"","x":810,"y":240,"wires":[["956e3f86.05c64","fd8753e7.9c755"]]},{"id":"956e3f86.05c64","type":"function","z":"2927dc06.684b3c","name":"For loop IP","func":"var str = msg.payload ;\nvar arraylength = str.length ;\nfor (var i = 0; i < 2; i++) {\n   msg.payload = str[i]['ip_client'];\n   global.set(\"IP[\"+i+\"]\", str[i]['ip_client']);\n   node.send([{\"nip\":str[i]['ip_client']}, {\"payload\":true} ,{\"payload\":true}  ]);\n   //node.send([null, null , {\"payload\":true}  ]);\n\n}\n\n\n","outputs":3,"noerr":0,"x":970,"y":240,"wires":[["f42fa1b3.439c4"],[],["e55ee87b.791988"]]},{"id":"fd8753e7.9c755","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":950,"y":380,"wires":[]},{"id":"f42fa1b3.439c4","type":"function","z":"2927dc06.684b3c","name":"Connect","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://'+msg.nip+':4840',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\n\nreturn msg;","outputs":1,"noerr":0,"x":1200,"y":120,"wires":[["f94bd92.89fe2a8","303be4c2.171104"]]},{"id":"62f71b0a.0ae30c","type":"join","z":"2927dc06.684b3c","name":"Stack msg","mode":"custom","build":"array","property":"payload","propertyType":"msg","key":"topic","joiner":"\\n","joinerType":"str","accumulate":false,"timeout":"4","count":"","reduceRight":false,"reduceExp":"","reduceInit":"","reduceInitType":"num","reduceFixup":"","x":1570,"y":120,"wires":[["243a3112.b2c086"]]},{"id":"243a3112.b2c086","type":"function","z":"2927dc06.684b3c","name":"Stockage Values","func":"var str2 = msg.payload ;\nvar arraylength2 = str2.length ;\nfor (var i = 0; i < arraylength2; i++) {\n   msg.payload = str2[i];\n   global.set(\"Value3[\"+i+\"]\", str2[i]);\n   //node.send([{\"payload\":str[i]['ip_client']}]);\n  // node.send([{\"payload\":RangeTemperaturesA[i]['ip_client']}]);\n\n}\n","outputs":1,"noerr":0,"x":1750,"y":120,"wires":[[]]},{"id":"e55ee87b.791988","type":"OpcUa-Item","z":"2927dc06.684b3c","item":"ns=2;s=unit/Varihm.VIHM_GEN_rFCourAffich","datatype":"Float","value":"","name":"Format","x":1200,"y":180,"wires":[["f94bd92.89fe2a8","53d61cb.e1b7d64"]]},{"id":"f94bd92.89fe2a8","type":"OpcUa-Client","z":"2927dc06.684b3c","endpoint":"df932693.6c33b8","action":"read","time":10,"timeUnit":"s","name":"","x":1380,"y":120,"wires":[["321b3dc1.c48bea"]]},{"id":"70510669.9a6d78","type":"inject","z":"2927dc06.684b3c","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":450,"y":460,"wires":[["3c6d910.ad699f"]]},{"id":"3c6d910.ad699f","type":"function","z":"2927dc06.684b3c","name":"","func":"//var arraylength = global.Value.length ;\nfor (var i = 0; i < 3; i++) {\nvar myfoo1 = global.get(\"IP[\"+i+\"]\"); \nvar myfoo2 = global.get(\"Value3[\"+i+\"]\"); \nmsg.topic = \"UPDATE `liste_param` SET `Var3`='\"+myfoo2+\"' WHERE `ip_client`='\"+myfoo1+\"'\";\nnode.send(msg);\n}\n","outputs":1,"noerr":0,"x":600,"y":460,"wires":[["8a2740ed.1eb908","8f44bc.bcf89b48"]]},{"id":"8a2740ed.1eb908","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":730,"y":440,"wires":[]},{"id":"8f44bc.bcf89b48","type":"mysql","z":"2927dc06.684b3c","mydb":"2319a0e1.5a665","name":"","x":730,"y":480,"wires":[[]]},{"id":"303be4c2.171104","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1410,"y":220,"wires":[]},{"id":"53d61cb.e1b7d64","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1410,"y":280,"wires":[]},{"id":"321b3dc1.c48bea","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1560,"y":200,"wires":[]},{"id":"2319a0e1.5a665","type":"MySQLdatabase","z":"","host":"localhost","port":"3306","db":"client","tz":""},{"id":"df932693.6c33b8","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://192.168.210.100:4840","secpol":"None","secmode":"NONE","login":false}]`

Thanks,

Lothdeorn, 2018-12-13:
I've manage to do something who is working.

It's working well. It just send wrong value if the opcua node can't connect to the machine. 

> [{"id":"2927dc06.684b3c","type":"tab","label":"Flow 3","disabled":false,"info":""},{"id":"4c4177c8.4a4bd8","type":"inject","z":"2927dc06.684b3c","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":470,"y":240,"wires":[["ad225a5c.1f762","b27879e.8e73208"]]},{"id":"ad225a5c.1f762","type":"function","z":"2927dc06.684b3c","name":"IP CLIENT","func":"msg.topic = \"SELECT ip_client FROM `liste_machines`\" ;\n\nreturn msg;","outputs":1,"noerr":0,"x":650,"y":240,"wires":[["5398a0b8.935ea8"]]},{"id":"5398a0b8.935ea8","type":"mysql","z":"2927dc06.684b3c","mydb":"2319a0e1.5a665","name":"","x":810,"y":240,"wires":[["956e3f86.05c64","fd8753e7.9c755"]]},{"id":"956e3f86.05c64","type":"function","z":"2927dc06.684b3c","name":"For loop IP","func":"var str = msg.payload ;\nvar arraylength = str.length \n\nasync function LoopIP(){\n  // node.send([{\"payload\":true}]);\n  for (var i = 0; i < arraylength; i++) {\n\n    global.set(\"IP[\"+i+\"]\", str[i]['ip_client']);\n    node.send([{\"nip\":str[i]['ip_client'], \"payload\":true}, null , null ]);\n    await sleep(200)\n    node.send([null , null , { \"payload\":true } ]);\n    await sleep(200)\n\n    }\n    node.send([null, { \"complete\":true }  , null ]);\n    \n}\n\nfunction sleep(ms){\n    return new Promise(resolve=>{\n        setTimeout(resolve,ms)\n    })\n}\n\nLoopIP()\n\n\n\n\n","outputs":3,"noerr":0,"x":970,"y":240,"wires":[["f42fa1b3.439c4"],["e935011.dfd6a8","62f71b0a.0ae30c"],["568394bd.3b59bc"]]},{"id":"fd8753e7.9c755","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":950,"y":380,"wires":[]},{"id":"f42fa1b3.439c4","type":"function","z":"2927dc06.684b3c","name":"Connect","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://'+msg.nip+':4840',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\n\nreturn msg;","outputs":1,"noerr":0,"x":1140,"y":120,"wires":[["f94bd92.89fe2a8","dd755e87.2d2aa8"]]},{"id":"62f71b0a.0ae30c","type":"join","z":"2927dc06.684b3c","name":"Stack msg","mode":"custom","build":"array","property":"payload","propertyType":"msg","key":"topic","joiner":"\\n","joinerType":"str","accumulate":false,"timeout":"","count":"","reduceRight":false,"reduceExp":"","reduceInit":"","reduceInitType":"num","reduceFixup":"","x":1550,"y":240,"wires":[["243a3112.b2c086","a7f9b3ee.4bdb08"]]},{"id":"243a3112.b2c086","type":"function","z":"2927dc06.684b3c","name":"Stockage Values","func":"var str2 = msg.payload ;\nvar arraylength2 = str2.length ;\nfor (var i = 0; i < arraylength2; i++) {\n   //msg.payload = str2[i];\n   global.set(\"Value3[\"+i+\"]\", str2[i]);\n   node.send([{\"payload\":str2[i]}]);\n  // node.send([{\"payload\":RangeTemperaturesA[i]['ip_client']}]);\n\n}\n","outputs":1,"noerr":0,"x":1770,"y":180,"wires":[["2d284ce4.05907c"]]},{"id":"f94bd92.89fe2a8","type":"OpcUa-Client","z":"2927dc06.684b3c","endpoint":"da97cddf.0b0648","action":"read","time":10,"timeUnit":"s","name":"","x":1340,"y":120,"wires":[["68328bae.6dca74","62f71b0a.0ae30c"]]},{"id":"70510669.9a6d78","type":"inject","z":"2927dc06.684b3c","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":470,"y":460,"wires":[["3c6d910.ad699f"]]},{"id":"3c6d910.ad699f","type":"function","z":"2927dc06.684b3c","name":"","func":"var arraylength = global.Value.length ;\nfor (var i = 0; i < 3; i++) {\nvar myfoo1 = global.get(\"IP[\"+i+\"]\"); \nvar myfoo2 = global.get(\"Value3[\"+i+\"]\"); \nmsg.topic = \"UPDATE `liste_param` SET `Var3`='\"+myfoo2+\"' WHERE `ip_client`='\"+myfoo1+\"'\";\nnode.send(msg);\n}\n","outputs":1,"noerr":0,"x":620,"y":460,"wires":[["8a2740ed.1eb908","8f44bc.bcf89b48"]]},{"id":"8a2740ed.1eb908","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":750,"y":440,"wires":[]},{"id":"8f44bc.bcf89b48","type":"mysql","z":"2927dc06.684b3c","mydb":"2319a0e1.5a665","name":"","x":750,"y":480,"wires":[[]]},{"id":"b27879e.8e73208","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":610,"y":320,"wires":[]},{"id":"568394bd.3b59bc","type":"OpcUa-Item","z":"2927dc06.684b3c","item":"ns=2;s=unit/Varihm3.VIHM_prdc_iiComptTotSacProdCour","datatype":"Int32","value":"","name":"NbSac","x":1170,"y":160,"wires":[["f94bd92.89fe2a8","9f152411.da03a"]]},{"id":"dd755e87.2d2aa8","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1290,"y":60,"wires":[]},{"id":"9f152411.da03a","type":"debug","z":"2927dc06.684b3c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1310,"y":180,"wires":[]},{"id":"68328bae.6dca74","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1530,"y":60,"wires":[]},{"id":"e935011.dfd6a8","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1190,"y":360,"wires":[]},{"id":"a [... cut ...] :1750,"y":260,"wires":[]},{"id":"2d284ce4.05907c","type":"debug","z":"2927dc06.684b3c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":1970,"y":220,"wires":[]},{"id":"2319a0e1.5a665","type":"MySQLdatabase","z":"","host":"localhost","port":"3306","db":"client","tz":""},{"id":"da97cddf.0b0648","type":"OpcUa-Endpoint","z":"2927dc06.684b3c","endpoint":"opc.tcp://TREL16087112:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

Lothdeorn, 2018-12-13:
It will be great to have the endpoint information in the msg object response from the opcua client : 

> : msg : Object
> object
> payload: 25
> _msgid: "d5c220f7.68777"
> topic: "ns=2;s=unit/Varihm3.VIHM_prdc_iiComptTotSacProdCour"
> datatype: "Int32"
> browseName: "NbSac"

Then we can know from which endpoint the value is coming.

mikakaraila, 2018-12-16:
Ack, bit busy time, but I look after couple of days...

closed 2021-06-10 by mikakaraila