mikakaraila/node-red-contrib-opcua #97: This node causes Node.JS module get-installed-path to stop working

TotallyInformation, 2019-01-22:
get-installed-path is used by node-red-contrib-uibuilder to find the install paths for front-end libraries that have been installed using npm.

If this node is installed, the sync version of the function `getInstalledPathSync` stops working. I've not tried the other version which returns a promise.

The issue lies at [line 130 of that module](https://github.com/tunnckoCoreLabs/get-installed-path/blob/master/src/index.js#L130):

    const modulePath = filePaths.find((filePath) => {
    ...

The find function is never executed.

If node-red-contrib-opcua is removed, the function starts working again.

To me, this implies that something that this node is doing is overwriting the native `Array.prototype.find` function? Everything else appears to be working as normal.

mikakaraila, 2019-02-11:
There is actually no exports and Array is not used so it could be overwrite it. I suppose it could come from some underlying dependency instead of directly node-red-contrib-opcua. Have to investigate more...

TotallyInformation, 2019-02-11:
Hi, thanks for the response. I don't actually use your node myself as I don't have the hardware but as you can see from the reference, one of the users of uibuilder reported it.

> extends core types (e.g extends Array.prototype with additional non-enumerable properties like .set)

Urgh! Not a good approach! Sounds like you are right. This illustrates exactly why extending the prototype of a native object isn't a good approach. Too easy to break something.

TotallyInformation, 2019-02-11:
Though, I can see that the collections package is actually at v5, you are using v3. Possibly this has been fixed in a later release?

mikakaraila, 2019-02-11:
Can you test it now? I am at work and I will have time perhaps later today?

mikakaraila, 2019-02-11:
OK, I updated collections and node-opcua to latest ones.

mikakaraila, 2019-02-14:
Has anybody tested new version?

closed 2019-02-17 by mikakaraila