mikakaraila/node-red-contrib-opcua #99: the server returns "BadCertificateUseNotAllowed (0x80180000)"

ilpaso, 2019-02-05:
I'm trying to connect to a Sinumerik OPC-UA server with node-red.

The node returns the error "BadCertificateUseNotAllowed (0x80180000)" and I do not find a solution.
After the connection attempt I can find a directory called pki/untrusted with my server certficate (.pem).

I'm able to connect to the server with the software UaExpert and at the first connection I set the certificate as trusted.

Any clue?
Thank you
regards

mikakaraila, 2019-02-06:
Can you run flow with verbose on? Command "node-red -v flow.json" to see what is happening...
For complete error (protocol) level you can first define "set DEBUG=ALL" on shell before starting node-red.

ilpaso, 2019-02-07:
thank you for your help!
this is the log:

> 7 Feb 13:26:10 - [info] [OpcUa-Client:a81cb69a.e1f038] Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2019-02-07T12:27:41.374Z
   requestHandle               /* IntegerId                        */: 3               0x3
   serviceResult               /* StatusCode                       */: BadCertificateUseNotAllowed (0x80180000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
    at process_request_callback (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:69:15)
    at ClientSecureChannelLayer._on_message_received (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:129:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-secure-channel/src/message_builder.js:461:22)
    at MessageBuilderBase._feed_messageChunk (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-transport/src/message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-transport/src/message_builder_base.js:53:14)
    at emitOne (events.js:116:13)
    at PacketAssembler.emit (events.js:211:7)
    at PacketAssembler.feed (/home/paso/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-packet-assembler/src/packet_assembler.js:107:14)
7 Feb 13:26:10 - [info] [OpcUa-Client:a81cb69a.e1f038] Client status: connection error
7 Feb 13:26:10 - [info] [OpcUa-Client:a81cb69a.e1f038] Client disconnected!

mikakaraila, 2019-02-10:
Have you checked your server certificate handling? Is it automatic or manual?
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&ved=2ahUKEwiP8PazlrHgAhVl6KYKHUegB7wQFjABegQICRAC&url=https%3A%2F%2Fsupport.industry.siemens.com%2Fcs%2Fattachments%2F109760282%2F840Dsl_828D_OPCUA_config_man_0818_en-US.pdf%3Fdownload%3Dtrue&usg=AOvVaw2Uiy7-hN4hwhsw28gcOfA9

Read page 24.

ilpaso, 2019-02-10:
Thank you Mika for your reply,
when I try to connect with node-red the server returns the error I wrote in my previous post and a pem file is created in pki/untrusted directory. The filename of this pem is the same I found in the server so I put the certificate as trusted at the server side in order to be sure I do not have problems at server side.
So I think it doesn't matter if certificate handling is automatic or manual.

Second: if I try to connect to server with UAExpert it returns a "untrusted certificate message" as shown in this image: http://documentation.unified-automation.com/uaexpert/1.4.1/html/certificate_validation.png
But in this case there is the button "Trust server certificate" and everything goes perfect.

In my opinion the problem is not at server side but I've to trust the server certificate at client side and I'm not able to do that. Maybe I've to move the pem file in any directory but I do not know where.

I've read a lot a documents or forum messages but the only interesting post is this: https://github.com/node-opcua/node-opcua/issues/549
I think it's a problem very similar to mine.

Regards

mikakaraila, 2019-02-10:
Did you test solution? I mean you can generate client certificate according those instructions.
Then server should accept it if the case is same.

ilpaso, 2019-02-10:
I didn't test the solution yet.
I thought that trusting the server certificate was a standard case and was enough move the pem file from the pki/untrusted directory to another directory in order to reproduce what is performed in the UAExpert software.

mikakaraila, 2019-02-11:
I expect that Siemens Server keeps track on those, so just copy of files is not enough.

mikakaraila, 2019-02-11:
Please test now with new version. There was wrong path into certificates in my code.

ilpaso, 2019-02-12:
Now it returns this error:
Error: cannot locate private key file /home/paso/.node-red/node_modules/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem

please see the double "node_modules" directory in the path! Maybe that is wrong.

If I move the "private_key.pem" and "client_selfsigned_cert_1024.pem" the new error goes but returns the bad old "BadCertificateUseNotAllowed (0x80180000)" again.

mikakaraila, 2019-02-14:
I rolled back path, but I suppose server should import client certificate. File copy is not enough.

mikakaraila, 2019-02-17:
There seems to be wrong certificate... I am fixing code. Will take some time.

mikakaraila, 2019-02-18:
Code will use .node-red/pki to store server certificate. It works I just tested it...
I will try to copy server certificate manually and check it will work.

ilpaso, 2019-02-19:
Thank you. I will test it when ready.

mikakaraila, 2019-02-20:
This one should be ok.

ilpaso, 2019-02-20:
I've updated to last release but the error "BadCertificateUseNotAllowed (0x80180000)" persists. Any clues?

mikakaraila, 2019-02-20:
Can you run it with security policy / mode None / none ? Client can read certificate and use it then locally?
Look additional parameters added to client node.

wei1224hf, 2019-02-25:
It's the client should set the EndpointDescription's MessageSecurityMode to NONE.

ilpaso, 2019-02-26:
Where can I set the MessageSecurityMode to NONE in the node-red-contrib-opcua setup?

mikakaraila, 2019-02-27:
Somebody wanted it to be removed (security reason), but I think it is still needed.
I will put it back, later today.

mikakaraila, 2019-03-11:
Reopen if still something...

closed 2019-03-11 by mikakaraila