mikakaraila/node-red-contrib-opcua #101: Path to certificates is wrong in 0.2.38

tomjax, 2019-02-18:
Cannot read:/home/xxx/.node-red/node_modules/node_modules/node-opcua-client/certificates/client_selfsigned_cert_1024.pem file.

"node_modules" comes twice

mikakaraila, 2019-02-18:
I know, fixing the original cause. Please wait...

mikakaraila, 2019-02-18:
Kind of solution coded, but bit clumsy. 
It will need first to connect with Security Policy & Mode as None.
It can be used to get server certificates.
Then certificate can be used to connect with wanted security settings.
Of course another way is to manually copy certificate pem file to local folder (under pki).
Any comments?

tomjax, 2019-02-18:
For me it worked ok to just edit the file:
/home/xxx/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js
and change the path from "/../../" to "/../../../"

mikakaraila, 2019-02-18:
It seems to work, but actually it is not using the real certificate. Client can read server certificate through endpoint. Look the latest code how it is done now.

tomjax, 2019-02-18:
Hi, now it works for me. Tomorrow I will test some more on some real OPCUA-servers.

mikakaraila, 2019-02-20:
No news? Expecting this works now...

closed 2019-02-20 by mikakaraila

tomjax, 2019-02-20:
Sorry for late answer, yes it works! :)