mikakaraila/node-red-contrib-opcua #102: Subscription does not work as expected [OPC Client with SUBSCRIBE]

GdaA3O, 2019-02-18:
Hello community

Let me explain :
I have a temperature variable in my Siemens API. It changes 10 times a second. I use "OPC Client" with the option "SUBSCRIBE" to make a subscription of this tag every 1 seconds. As a result, I get the 10 different states of my temperature every second (every second, i have 10 msg.payload).

The good behavior should be: 
if my temperature to move after 1 second then I receive the last value.
This is what I get with a program written with the python librairy- opcua

Thank you for your explanations

mikakaraila, 2019-02-18:
Subscription is to get value as it is changing. There is some option to use deadband.
BUT I suppose you should use simple read with injecting with wanted interval to read value.

GdaA3O, 2019-02-18:
OK I understand that in your node "OPC UA Client" with the option "SUBSCRIBE" you retrieve all data changes that you distribute on the time base parameter's. I have good?

However, this has nothing to do with the real subscription of datas to an OPC Server. It is not interesting to have all the values of the tags ALONE the last one interests us ..
There is ambiguity. 

The discussion is open ...

Thank you

mikakaraila, 2019-02-18:
There are other parameters in subscription: publishingInterval and samplingInterval those could be perhaps used to get wanted values out.
But those are not possible to modify at the moment through node.

abrasat, 2019-03-05:
In what release do you plan to expose the subscription parameters publishingInterval and samplingInterval ?
What exactly do you mean with deadband, where can this be set and how is it expected to work?
And one question about the usage of the node red OPC UA client with "Subscribe": is it necessary to call it cyclically ? Is it not enough to subscribe once, and then OPC UA client output will be triggered as soon as a new value for the subscribed item is received from the OPC UA server?

mikakaraila, 2019-03-05:
Quick and dirty way is to implement new node with needed parameters. 
Subscribe is needed to inject only once then client will emit new value to output.
I will add new node today / tomorrow... not so big work.

abrasat, 2019-03-06:
Ok, thx. Is the interval value used for the publishingInterval or for the samplingInterval parameter?

mikakaraila, 2019-03-06:
Ok, I will check it at home later today. But anyway I suppose this was feature you were asking?

abrasat, 2019-03-06:
Not sure about that...
I was actually looking for the possibility to explicitly set on the OPC UA client side the parameters publishingInterval and samplingInterval when generating a subscription for a (or several) monitored item(s). An OPC UA session should be able to handle several subscriptions, this is not possible to be achieved with the actual implementation, as each "OPC UA Client" node opens a new session (with some automatically generated session name - not possible to be set from outside), and has the action type inside it. The "action" (Subscribe, Monitor and so on) would better belong to the "OPC UA Item" node. Or maybe you should create specialized nodes for each kind of action. With the actual implementation I dont think that it is possible to have several subscriptions sharing the same session.

mikakaraila, 2019-03-06:
Ok, somebody will do the fork or will create new nodes. Original design was to make easy way to subscribe variables (auto create session inside).

abrasat, 2019-03-06:
Regarding the actual implementation, could you please just clarify how are set the publishingInterval and the samplingInterval parameters when using the "monitor" action?

mikakaraila, 2019-03-06:
Some clarifications:
1)     SamplingInterval can be given in msg.payload
var interval = convertAndCheckInterval(msg.payload);
verbose_log(msg.topic + " samplingInterval " + interval);

2) PublishingInterval can be given in Client node, but there was mistake it was not given as option.
    Fix is below and in next version.    
newSubscription = new opcua.ClientSubscription(node.session, {requestedPublishingInterval: parameters});

closed 2019-03-06 by mikakaraila

abrasat, 2019-03-07:
Thx for the clarification.
Is the samplingInterval given as Date object?
Is it to possible to set also the action type for the OPC UA Client node via msg properties?