mikakaraila/node-red-contrib-opcua #103: Not working with node v8.11.1 and nodered 0.19.4

thecalle, 2019-02-20:
Environment: Raspbian
node v.8.11.1
nodered 0.19.4
Running npm install node-red-contrib-opcua give some warnings, but Done at the end. Accessing the Palette, version 0.2.39 is installed, but nothing is working. 6/6 nodes are not in use and I got a list of object Object with an exclamation mark icon at the end.

mikakaraila, 2019-02-20:
Some debug output would help to start fixing...

thecalle, 2019-02-20:
Tons of dependencies missing.

Last one: [warn] [node-red-contrib-opcua/OpcUa-Client] TypeError: Cannot read property 'encode' of undefined

The client is not working..

mikakaraila, 2019-02-20:
Most probably something in your environment... I do not have Raspian at the moment available to test this.
Can you study more or try to install again and test also command line installation?

thecalle, 2019-02-21:
I'm investigating on a brand new image about the problem. At the beginning, node version present in Rasbian Stretch is v.4.8.2. opcua install aborts due to requesting 6.X or more. If you update node, you jump to 8.11.1 but apt installer removes about 40 node packages. I'm trying to upgrade to 6.X. I will update you ASAP

thecalle, 2019-02-21:
Fresh install, moved to node 6.10 manually, opcua ok. OpcUa-Endpoint is greyed an not present in Nodered left pane nodes. Here we can find an easy script to choose the node version manually and install quickly: https://github.com/audstanley/NodeJs-Raspberry-Pi

mikakaraila, 2019-02-21:
BTW opcua-endpoint is config node... part of opcua-client.

closed 2019-03-06 by mikakaraila