mikakaraila/node-red-contrib-opcua #106: Unsubscribe doesn't work

m-obi, 2019-03-07:
When I want to Unsubscribe, it doesn't work. I'm writing unsubscribe in msg.action like in the example.
But then an error occured with "Item not monitored...". But I subscribed before.

[image]

mikakaraila, 2019-03-08:
Most probably bug as I changed set-handling package.

mikakaraila, 2019-03-11:
I cleaned up code and reimplemented unscubscribe Set -> Map, simplified.
I will publish code after work later today.

mikakaraila, 2019-03-11:
Please test with new version.

m-obi, 2019-03-11:
Perfect. It's working. There is only an issue with the interval for now. It's set to 100 ms even if I set to 2 sec.

mikakaraila, 2019-03-11:
I actually change code a bit default value is 100ms and payload (number) can be used to set interval. How did you try to set it in your flow?

m-obi, 2019-03-11:
It's working with payload, but only with data types which are listed. Not with an array. When will this possible?

m-obi, 2019-03-11:
Those with array is publishing when the value changed.

mikakaraila, 2019-03-11:
Implementing Array support is bit more work. I have done for our system.
I will close this issue now.

closed 2019-03-11 by mikakaraila