mikakaraila/node-red-contrib-opcua #107: Connect and reconnect issue - Timeout.......waiting for response for PublishRequest

japloi, 2019-03-22:
Hi,

I having trouble staying connected and subscribing to the OPC recently, where i keep getting "Timeout.....waiting for response for PublishRequest" after a while it will automatically reconnected and starts subscribing again.
[image]

mikakaraila, 2019-03-24:
What is server? Do you a lot of variables?

japloi, 2019-03-25:
Thanks for the reply?
I'm using the latest opc ua but still have disconnect and reconnect. On the opcua client, I'm connecting to an IBH OPC server.

mikakaraila, 2019-03-26:
Ack, I will test with demo server later. No time due traveling at the moment.

japloi, 2019-03-26:
Thanks you, safe travels

japloi, 2019-03-28:
Hi Mikakaraila, have you have the time to view this yet?

mikakaraila, 2019-03-28:
Just back at work, demo server installed.

japloi, 2019-03-28:
Thanks, Im in work aswell, if you need me to check anything from my side, let me know, also i have checked some of the issues on here and i have a similar issue as  #89 as i have to redeploy to get the connection back up working again and it runs for 2-3mins and it timeout again. 
I then tried option 2 where i leave it without redeploy and it will start backup itself after 3mins of running it time out for 8mins and reconnect. 
As the data are stored on chronograph i see that the its disconnects and connect consistantly.
Please see screenshot below
[image]
[image]

mikakaraila, 2019-03-28:
Can you provide some test configuration for server?
Did you install node-opcua v0.7.1

japloi, 2019-03-28:
What kind of test configuration do you need? and you be more specific please.
I have installed node-opcua v0.7.0 i didnt know that its has been updated, i will update now

japloi, 2019-03-28:
Updated and tested 0.7.1, still have the same issue.

japloi, 2019-03-28:
Hi Mikakaraila, Just a quick update,
It could be the security settings in node red that causing the drop out as i remove the Security and set to none, there is no timeout.
As i have another OPCUA softing server running with no security and certificates and it doesn't dropout/timeout but the IBH has security and an certificate and it timeout. 
Once i removed it, it works with no timeout.
Regards
Loi

mikakaraila, 2019-03-28:
OK, then this could lead us to check how node-opcua uses timestamp (accuracy) in certificate checking with server...

japloi, 2019-03-28:
Will it be easy to fix? Or is there a work around?

mikakaraila, 2019-03-31:
This will take some time to find what will cause this? Have you checked all connected computer are using same time and time zone settings? IBH Server documentation says that with security settings they must be same.

japloi, 2019-04-02:
Hi Makakaraila,Thenks for the feedback, after reviewing the times on both system, the companies pc is the companies server and the IBH was on a different server. So it was out of sync thus causing the Timeout.
after i sync the IBH to the our server and ran your opcua, the timeout is gone. Thanks you for your help.

Just a quick question, What is the fastest speed i can set on the client? that i can get data from the IBH server? and whats does the latency depends on? i trying to measure the latancy of the data.

Thanks again

mikakaraila, 2019-04-02:
Here is one good article:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=11&cad=rja&uact=8&ved=2ahUKEwiby4D9v7HhAhVFw4sKHb49B6w4ChAWMAB6BAgDEAI&url=https%3A%2F%2Fwww.thinkmind.org%2Fdownload.php%3Farticleid%3Dintelli_2015_9_40_95011&usg=AOvVaw2fdFERT_jXjG3iS3xJUaxj
But in many cases it depends on chain: PLC - OPC UA Server - OPC UA Client
First step is to check PLC execution cycle, then do you know how data is bind to server.
After that you can use subscription and monitoredItem parameters to fine tune transfer.

closed 2019-04-03 by japloi