mikakaraila/node-red-contrib-opcua #108: OPCUA Browser issue

greatdsa, 2019-04-11:
Browser_contrib_ui_flow.json is not working for me.. 

I need a browser to implement which browses all the available items and shows them on dashboard. then I should be able to select among the available items to visualize the desired values of the items.

mikakaraila, 2019-04-16:
Aah, I think one modification broke this...

greatdsa, 2019-04-16:
Can you pelase help me to solve this issue?

mikakaraila, 2019-04-16:
Traveling today, let's see if I have some time tomorrow.

mikakaraila, 2019-04-17:
Seems to work, now I will cleanup and then commit updated example and fix soon.

greatdsa, 2019-04-17:
I am using the fixed one now. There are no errors, no crash but I am not getting any data. I am using the Prosys simulation server. I haven't anything other than the server address. When I click on Browse, I get a notification of Browsing and tehn nothing happens. No item is retrieved.

mikakaraila, 2019-04-17:
Did you update endpoint? I tested it also with Prosys simulation server and it worked.

Most probably example contains now my hostname.

greatdsa, 2019-04-18:
I have changed the endpoint to my prosys simulation server endpoint and it is not getting any item. When I browse, it shows "Browsing..." and then nothing happens

mikakaraila, 2019-04-18:
I will get this:
[image]

And if I select simulation then click Browse:
[image]

mikakaraila, 2019-04-18:
Can you provide debug output?
Start: "node-red -v Browse.json"

greatdsa, 2019-04-18:
[image]

greatdsa, 2019-04-18:
I have only changed the endpoint. Everything else is the same. I am also using Prosys Simulation Server. I am not getting any item.

greatdsa, 2019-04-18:
[image]
[image]
[image]

greatdsa, 2019-04-24:
@mikakaraila, Do I have to change anything other than the endpoint? 
Because I haven't changed anything except the endpoint.

mikakaraila, 2019-04-24:
I would like to get console output not the browser debug side... Then I can see more what is happening.

greatdsa, 2019-04-24:
It is working now. 

Actually I was using the wrong file. You can close the issue now. Thanks a lot.

closed 2019-04-25 by mikakaraila