mikakaraila/node-red-contrib-opcua #110: Error: get-installed-path: module not found "node-opcua-server" in path

RMutharaju, 2019-05-20:
Hi,

I have installed Node-RED 0.20.3 and node-red-contrib-opcua 0.2.45. I have created a flow as below and deployed the same..

[{"id":"aaab478b.c08c28","type":"inject","z":"caeb8ff7.35b44","name":"restartOPCUAServer","topic":"","payload":"{\"opcuaCommand\":\"restartOPCUAServer\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":280,"y":380,"wires":[["4171b43e.a9959c"]]}]

In debug tab I see 
msg: error
"Error: get-installed-path: module not found "node-opcua-server" in path /home/nodered/node-red/node-red-contrib-opcua/node-opcua-server"

Please let me know in case any further details are required. Any ideas on how to fix this issue or future plans regarding the same.

Thanks for the help.
Best regards,
Rakesh

tot92, 2019-06-06:
Hi Rakesh,

I ran into the same issue.

Fix:
in (/home/nodered/node-red/node-red-contrib-opcua) /opcua/104-opcuaserver.js, line 85:
replace 
path.join(__dirname, '..'),
with
path.join(__dirname, '../..'),

Explanation:
It cannot find the "node-opcua-server" module because it looks in the wrong directory!
It must look in the grandparent directory instead of the parent dir...

Hope that helps...

mikakaraila, 2019-06-11:
Modified code according your input, thanks.

closed 2019-06-11 by mikakaraila