mikakaraila/node-red-contrib-opcua #113: OPC UA connection error: Backoff aborted

johnwayne111, 2019-06-07:
Hi,

I am trying to set my IoT2040 to be client (using node-red-contrib-opcua), with the OPC UA simulator "Prosys" as a server and I receive this error: "OPC UA connection error: Backoff aborted." and the node "OPC UA Client" keeps a connecting status. 
Moreover, during the deploy, the status of my node "OPC UA Client" is closed with a red flag.

[image]

[image]

I'm sure that the server endpoint address on "OPC UA Client" node is correct because I checked it with another client (UaExpert).

I tried also with another OPC UA server simulator "KEPServerEX" as a server and the result is another error: "OPC UA connection error: connect ECONNREFUSED". I tested also it with UaExpert and, like the previous example, it works using the same endpoint address that I set in "OPC UA Client" node.

Any idea?

Thanks in advance

mikakaraila, 2019-06-07:
Did you accept node-red client on KepServer?

johnwayne111, 2019-06-07:
No. when I try to establish a communication with KEPServerEx as a server, I receive this message: "OPC UA connection error: connect ECONNREFUSED".

mikakaraila, 2019-06-07:
Yes error comes because you haven't accepted / white listed client on server side. Look through server manual...

johnwayne111, 2019-06-10:
It's very strange because I tried also the same sample in Node-RED that runs on my host with windows 10 and it works. Unfortunately, it keeps the problems in Node-RED that runs in my IOT2040: they support the same Node-RED version.
Is "white list client" present in both server? Because I didn't find it in KepserverEx and Prosys.
Moreover, I don't understand why the servers should accept the connection with my host and refuse the connection with IOT2040, using the same Node-RED.

johnwayne111, 2019-06-10:
Probably I think I found this white list in Kepserverex because now the message on debug tab is changed: now is "OPC UA connection error: Backoff aborted" the same of the other case with Prosys.

mikakaraila, 2020-05-09:
Updated node-opcua should solve this.

closed 2020-05-09 by mikakaraila