mikakaraila/node-red-contrib-opcua #114: Update to node-opcua@2.0.0?

bruegth, 2019-07-04:
@mikakaraila Would you update node-opcua to version 2.0.0 ?

mikakaraila, 2019-07-04:
I have done it for our server and client.
But how to use it with node-red? 
Build will produce JavaScript of course.

mikakaraila, 2019-09-16:
I updated just to latest javascript node-opcua v0.7.4

lal12, 2020-02-12:
I also would with for an update to 2.0.0. Since the last versions, my build is broken since node-opcua@0.7.4 requires a native dependency which has to be compiled. 

> But how to use it with node-red?
> Build will produce JavaScript of course.
I am not sure if I understand right... However there is no need to care about the typescript part of node-opcua@2.0.0, since the compiled js version is also included, there are just some (AFAIK minor) api changes.

mikakaraila, 2020-02-13:
It needs only some minor re-factoring to TypeScript (ts) and then build + use as Javascript (js).
I will look it as own node-red-contrib-opcua-ts perhaps or node-red-contrib-ts-opcua
I will check others and ask from people how to name contrib with TypeScript.

mikakaraila, 2020-02-13:
Or just same package name with major update. This will take some time.

lal12, 2020-02-17:
I looked into it. It seems that just like 10 lines need very minor changes and one bigger change regarding subscription.monitor(...), where some code needs to be rearranged. I will try those changes tomorrow and if it works I will do a PR.

mikakaraila, 2020-02-18:
Great thanks, busy with standard working.

mikakaraila, 2020-02-18:
Ok, this is not yet complete. We need to compile ts files to js and add devDependencies to package.json

bruegth, 2020-05-26:
@mikakaraila This Issue can be closed I think, nor?

mikakaraila, 2020-05-26:
Yes I am still using latest node-opcua through JavaScript.

But this is ok to close

closed 2020-08-04 by mikakaraila