mikakaraila/node-red-contrib-opcua #115: emitting an error causes Node-Red to crash

nospam2000, 2019-07-10:
see my comment in #63:
The implementation of node_error() causes Node-Red to crash
```
function node_error(err) {
  node.error(err, err);
}
```

The second parameter needs to be the msg and not a string. Otherwise this causes Node-Red to completely crash when you use a "catch" node. 

See https://github.com/node-red/node-red/issues/2228 for the problem this causes

See also the documentation in https://nodered.org/docs/user-guide/writing-functions:
_Handling errors_
_If the function encounters an error that should halt the current flow, it should return nothing. To trigger a Catch node on the same tab, the function should call node.error **with the original message as a second argument**:_

`node.error("hit an error", msg);
`

mikakaraila, 2019-07-13:
Ack I will fix next week

mikakaraila, 2019-07-28:
Fixed, new version done.

commit f52e6af2 (2019-07-28): Fixed #115 node.error() function first parameter
closed 2019-07-28 by mikakaraila