mikakaraila/node-red-contrib-opcua #116: OPC UA connection error: Invalid endpoint url

RMutharaju, 2019-07-12:
HI,

I'm using OPCUA version 0.2.45

Environment:
Welcome to Node-RED
===================

12 Jul 08:41:59 - [info] Node-RED version: v0.20.3
12 Jul 08:41:59 - [info] Node.js  version: v8.15.1
12 Jul 08:41:59 - [info] Linux 4.14.87-rt49+ x64 LE
12 Jul 08:42:01 - [info] Loading palette nodes

**console logs:**
12 Jul 08:42:29 - [info] Starting flows
12 Jul 08:42:29 - [info] Started flows
12 Jul 08:42:29 - [info] Server now running at http://127.0.0.1:1880/
12 Jul 08:45:38 - [info] Stopping flows
12 Jul 08:45:38 - [info] Stopped flows
12 Jul 08:45:38 - [info] Starting flows
12 Jul 08:45:39 - [info] [OpcUa-Client:test]    No certificate used.
12 Jul 08:45:39 - [error] [OpcUa-Client:test] test OPC UA connection error: Invalid endpoint url
12 Jul 08:45:39 - [info] Started flows
12 Jul 08:45:39 - [red] Uncaught Exception:
12 Jul 08:45:39 - Error: Callback was already called.
    at /home/nodered/node-red/node-red-contrib-opcua/node_modules/async/dist/async.js:966:32
    at /home/nodered/node-red/node-red-contrib-opcua/node_modules/async/dist/async.js:3885:13
    at Timeout.delayer [as _onTimeout] (/home/nodered/node-red/node_modules/delayed/delayed.js:27:10)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
iii: finished entrypoint.sh

RMutharaju, 2019-07-12:
Flow used:
[opcua crash.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/3385787/opcua.crash.txt)

mikakaraila, 2019-07-13:
Ack should not crash.

mikakaraila, 2019-07-28:
See #115 and use catch node.

closed 2019-07-28 by mikakaraila

RMutharaju, 2019-09-03:
Hi,

The latest version of OPC-UA 0.2.47 and usage of Pure node.error(err) to catch node, this issue is reproducible in Node-RED 0.20.7.