mikakaraila/node-red-contrib-opcua #117: Limitation of parallel connections

danielpfeiferatadamos, 2019-08-05:
What are the steps to reproduce?
I use node-red (gui) and node-red-contrib-opcua with several data-point-items nodes and one opc-ua client node.
The server is provided by a third-party-company.
This was working fine in the past.

What happens?
The third-party-company limit parallel connections to server to 4 per client.
So I tried to limit the number of connections to 4 too, but this is not working.

What do you expect to happen?
I would expect some property will limit the number of connections for client via node-red settings. I tried different, no one have have any effect.

mikakaraila, 2019-08-05:
Check first with UaExpert server diagnostics then session and subscription diagnostics. 
Any error counter increasing?

danielpfeiferatadamos, 2019-08-05:
Yes, in UaExpert e.g. the RejectedSessionCount is increasing.
And in node-red log I get following:

5 Aug 19:13:03 - [info] [OpcUa-Client:Server] Client status: connection error
5 Aug 19:13:03 - [info] [OpcUa-Client:Server] Client status: session error
5 Aug 19:13:03 - [error] [OpcUa-Client:Server] Server OPC UA connection error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2019-08-05T17:13:02.669Z
   requestHandle               /* IntegerId                        */: 4               0x4
   serviceResult               /* StatusCode                       */: BadTooManySessions (0x80560000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
5 Aug 19:13:03 - [info] [OpcUa-Client:Server] Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2019-08-05T17:13:02.669Z
   requestHandle               /* IntegerId                        */: 4               0x4
   serviceResult               /* StatusCode                       */: BadTooManySessions (0x80560000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
    at process_request_callback (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:69:15)
    at ClientSecureChannelLayer._on_message_received (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:129:5)
    at MessageBuilder.emit (events.js:203:13)
    at MessageBuilder._decode_message_body (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\src\message_builder.js:462:22)
    at MessageBuilderBase._feed_messageChunk (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:53:14)
    at PacketAssembler.emit (events.js:203:13)
    at PacketAssembler.feed (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-packet-assembler\src\packet_assembler.js:107:14)
    at MessageBuilderBase.feed (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:159:30)
    at ClientSecureChannelLayer._on_receive_message_chunk (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:876:25)

I don't have the possibility to change the server-side, only my client.
The server is provided by a 3rd-part-company.

mikakaraila, 2019-08-05:
Ok it says: Too many sessions

Now we have to check is there session leak or really too many sessions open at the same time.

danielpfeiferatadamos, 2019-08-05:
The setup was running several days for hours without problem, then, without changes on client-side, they make the changes on server-side and the error-messages come right at the start-up. 
So, I'm pretty sure the client try to subscribe with "many" sessions at the same time.

mikakaraila, 2019-08-06:
Most probably client was using old sessions and as server was updated reconnect created new sessions.

danielpfeiferatadamos, 2019-08-06:
To clarify, I start node-red to create data and stop it after some time. This procedure for several days, for some hours.
Then one day I get the error message for two days and I asked the server provider, they told me they have limit the number of connections.

danielpfeiferatadamos, 2019-08-06:
Btw.
Perhaps I cut off the stacktrace a little bit early:
...    at MessageBuilderBase._feed_messageChunk (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:196:18)
    at PacketAssembler.<anonymous> (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:53:14)
    at PacketAssembler.emit (events.js:203:13)
    at PacketAssembler.feed (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-packet-assembler\src\packet_assembler.js:107:14)
    at MessageBuilderBase.feed (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\src\message_builder_base.js:159:30)
    at ClientSecureChannelLayer._on_receive_message_chunk (C:\Program Files\nodejs\versions\v12.6.0\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\src\client\client_secure_channel_layer.js:876:25)
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client status: connection error
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client disconnected!
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client disconnected!
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client disconnected!
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client disconnected!
6 Aug 09:51:43 - [info] [OpcUa-Client:Server] Client disconnected!
...

The fifth connection seems to cause the problem (limit number of connections) and the four already established connections will be closed. The retry ends with same result.

mikakaraila, 2019-08-06:
Ok reconnect strategy is too fast, it will try to connect and each time it creates new session that server doesn't close.

danielpfeiferatadamos, 2019-08-06:
Can I change this?

mikakaraila, 2019-08-06:
You can try, edit 102-opcuaclient.js and try with longer initialDelay and longer maxDelay.

 connectionOption.connectionStrategy = {
      maxRetry: 10512000, // 10 years should be enough. No infinite parameter for backoff.
      initialDelay: 2000,
      maxDelay: 30 * 1000
    };

mikakaraila, 2019-09-03:
Did you test this? Results...

danielpfeiferatadamos, 2019-09-04:
Sorry, out of mind ...
I did not test it, because the provider seems to change it again.
Thank you for your support.

mikakaraila, 2019-09-16:
OK server capacity problem.

closed 2019-09-16 by mikakaraila

danielpfeiferatadamos, 2019-09-16:
> OK server capacity problem.

Yes, but I'm pretty sure this would be a nice feature. Because can be a problem (dangerous) to open too much connections, IMO.

mikakaraila, 2019-09-16:
Client uses only 2 sessions: one for subscriptions and then other one for keep-alive.
So it is not exhausting connections. But if you add more client nodes then each one uses those 2 sessions.