mikakaraila/node-red-contrib-opcua #118: OPC UA "Error:Callback was already called." - crashes Node-RED 0.20.7

RMutharaju, 2019-09-03:
Hi Mika,

Reference Issue Id: 116

I have used Node-RED 0.20.7 and opcua v0.2.47. I have used the attached flow, when I deploy I get a Node-RED crash.
[flows (3).txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/3567924/flows.3.txt)

Welcome to Node-RED
===================

3 Sep 04:03:52 - [info] Node-RED version: v0.20.7
3 Sep 04:03:52 - [info] Node.js  version: v8.15.1
3 Sep 04:03:52 - [info] Linux 4.14.87-rt49+ x64 LE
3 Sep 04:03:53 - [info] Loading palette nodes
3 Sep 04:03:56 - [warn] rpi-gpio : Raspberry Pi specific node set inactive
3 Sep 04:03:56 - [warn] rpi-gpio : Cannot find Pi RPi.GPIO python library
Setting-up gobal tunnel with environment http_proxy.
3 Sep 04:04:06 - [info] Dashboard version 2.14.0 started at /ui
3 Sep 04:04:10 - [info] Settings file  : /home/nodered/sim-edge-settings.js
3 Sep 04:04:10 - [info] Context store  : 'default' [module=memory]
3 Sep 04:04:10 - [info] User directory : /home/nodered/node-red-home/node-red-flows
3 Sep 04:04:10 - [warn] No active project : using default flows file
3 Sep 04:04:10 - [info] Flows file     : /home/nodered/node-red-home/node-red-flows/flows.json
3 Sep 04:04:10 - [info] Creating new flow file
3 Sep 04:04:10 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

3 Sep 04:04:10 - [info] Server now running at http://127.0.0.1:1880/
3 Sep 04:04:10 - [info] Starting flows
3 Sep 04:04:10 - [info] Started flows
3 Sep 04:15:30 - [info] Stopping flows
3 Sep 04:15:30 - [info] Stopped flows
3 Sep 04:15:30 - [info] Starting flows
3 Sep 04:15:30 - [info] [OpcUa-Client:test]     No certificate used.
3 Sep 04:15:30 - [error] [OpcUa-Client:test] test OPC UA connection error: Invalid endpoint url
3 Sep 04:15:30 - [info] Started flows
3 Sep 04:15:31 - [red] Uncaught Exception:
3 Sep 04:15:31 - Error: Callback was already called.
    at /home/nodered/node-red/node-red-contrib-opcua/node_modules/async/dist/async.js:966:32
    at /home/nodered/node-red/node-red-contrib-opcua/node_modules/async/dist/async.js:3885:13
    at Timeout.delayer [as _onTimeout] (/home/nodered/node-red/node_modules/delayed/delayed.js:27:10)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
 
As suggested, I have used a catch node but the node.error(err) is not getting caught. Maybe that is not the root cause. Is there any way I can get more logs for this opcua-client node?

mikakaraila, 2019-09-03:
Start with -v option:
node-red -v test_flow.json

RMutharaju, 2019-09-05:
Here is the log with node-red -v

root@sim-edge-213:~# node-red -v
5 Sep 08:31:20 - [info]

Welcome to Node-RED
===================

5 Sep 08:31:20 - [info] Node-RED version: v0.20.7
5 Sep 08:31:20 - [info] Node.js  version: v10.16.3
5 Sep 08:31:20 - [info] Linux 4.14.103-rt55 x64 LE
5 Sep 08:31:21 - [info] Loading palette nodes
5 Sep 08:31:28 - [warn] rpi-gpio : Raspberry Pi specific node set inactive
5 Sep 08:31:28 - [warn] rpi-gpio : Cannot find Pi RPi.GPIO python library
5 Sep 08:31:29 - [info] Settings file  : /root/.node-red/settings.js
5 Sep 08:31:29 - [info] Context store  : 'default' [module=memory]
5 Sep 08:31:29 - [info] User directory : /root/.node-red
5 Sep 08:31:29 - [warn] Projects disabled : editorTheme.projects.enabled=false
5 Sep 08:31:29 - [info] Flows file     : /root/.node-red/flows_sim-edge-213.json
5 Sep 08:31:29 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

5 Sep 08:31:29 - [info] Starting flows
5 Sep 08:31:29 - [info] Started flows
5 Sep 08:31:29 - [info] Server now running at http://127.0.0.1:1880/
5 Sep 08:43:35 - [info] Stopping flows
5 Sep 08:43:35 - [info] Stopped flows
5 Sep 08:43:35 - [info] Starting flows
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921]   No certificate used.
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] [object Object]
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] [object Object]
5 Sep 08:43:35 - [warn] [OpcUa-Client:c10fb1d4.f4921] OpcUaClientNode: create Client ...
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] [object Object]
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] Client status: create client
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] async series - connecting
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] Client status: connecting
5 Sep 08:43:35 - [error] [OpcUa-Client:c10fb1d4.f4921]  OPC UA connection error: Invalid endpoint url
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] Error: Invalid endpoint url
    at parseEndpointUrl (/root/.node-red/node_modules/node-opcua-transport/src/tools.js:65:15)
    at ClientTCP_transport.connect (/root/.node-red/node_modules/node-opcua-transport/src/client_tcp_transport.js:144                                                                                                                        :16)
    at _connect (/root/.node-red/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:695                                                                                                                        :23)
    at FunctionCall.doCall_ (/root/.node-red/node_modules/backoff/lib/function_call.js:156:20)
    at FunctionCall.start (/root/.node-red/node_modules/backoff/lib/function_call.js:145:10)
    at _establish_connection (/root/.node-red/node_modules/node-opcua-secure-channel/src/client/client_secure_channel                                                                                                                        _layer.js:788:21)
    at ClientSecureChannelLayer.create (/root/.node-red/node_modules/node-opcua-secure-channel/src/client/client_secu                                                                                                                        re_channel_layer.js:792:5)
    at /root/.node-red/node_modules/node-opcua-client/src/client_base.js:390:27
    at /root/.node-red/node_modules/async/dist/async.js:3880:24
    at replenish (/root/.node-red/node_modules/async/dist/async.js:1011:17)
5 Sep 08:43:35 - [info] [OpcUa-Client:c10fb1d4.f4921] Client status: connection error
5 Sep 08:43:35 - [info] Started flows
5 Sep 08:43:35 - [red] Uncaught Exception:
5 Sep 08:43:35 - Error: Callback was already called.
    at /root/.node-red/node_modules/async/dist/async.js:966:32
    at /root/.node-red/node_modules/async/dist/async.js:3885:13
    at Timeout.delayer [as _onTimeout] (/root/.node-red/node_modules/delayed/delayed.js:27:10)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)

mikakaraila, 2019-09-05:
Please attach your flow here

RMutharaju, 2019-09-06:
[flows(2).txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/3581974/flows.2.txt)

I hereby attach the flows(2) (rename the extension as .json), connecting to OpcUa server with endpoint ( 132.186.90.176, securitypolicy: None, securityMode: None, use credentials: unchecked).

The endpoint url could be invalid, but I expect it should not lead to a Node-RED crash. The only way to get it working again would be to restart the Node-RED server which seems not a good solution.

RMutharaju, 2019-09-16:
Any help regarding this issue?

mikakaraila, 2019-09-16:
Just came back from US. I will take a look later today.

mikakaraila, 2019-09-16:
OK, found it & fixed. I will commit, push & publish new version.

mikakaraila, 2019-09-16:
As node-red starts then it will report: invalid endpoint ip-address and if you deploy then connection error.

closed 2019-09-16 by mikakaraila