mikakaraila/node-red-contrib-opcua #119: Conversion of MQTT protocol to OPC UA

smalhao, 2019-09-16:
Hello,

I've been working on converting MQTT topics to OPC UA for us, however I can't configure the OPC UA Server node, what I want is to put the pc where Node Red is installed as OPC UA Server, I configure to write the OPC UA Item to OPC UA Server but then when I inject it I don't add the node to the Server. My code is as follows:

`[{"id":"1f52f0ed.c5688f","type":"inject","z":"e2fa37a4.73a4d8","name":"","topic":"ns=2;s=LM35;Datatype=Float","payload":"","payloadType":"num","repeat":"","crontab":"","once":true,"onceDelay":0.1,"x":270,"y":620,"wires":[["870a1e1e.cd6b4"]]}]`

mikakaraila, 2019-09-16:
I have done it without node-red. Use mini-server and then mqtt package. Quite easy to write.

smalhao, 2019-09-16:
Mini server?Can you explain in other words?

mikakaraila, 2019-09-16:
I checked your flow: addVariable (no space in command) then datatype=Float (no camelcase) and finally longer wait needed like 5 seconds so server has started before adding new variables.
I added check for datatype so next version contains that check and I updated node-opcua to v0.7.4

closed 2019-09-16 by mikakaraila