mikakaraila/node-red-contrib-opcua #121: Error with OPC UA nodes connectivity

piyush-gupt, 2019-10-09:
I'm currently developing some dashboards where we have to fetch data from OPC UA server. Although I'm able to establish the connection, I'm getting the following error in receiving data from some specific nodes in the OPC UA server.
#getConstructor : cannot find constructor for expandedId ns=2; i=15004
#getConstructor : cannot find constructor for expandedId ns=2; i=15004

I'm expecting an array of structures giving x,y&z coordinate values. Since the array has 2 elements, I think I'm getting this error twice.

I also talked to OPC UA server provider and got the following info. The OPC UA server is based on 62541 OPC standard. The Id 15004 refers to the structure which is not getting deserialized in node red. OPC UA server uses binary encoding to encode this structure. Hence I'm getting the values as an array of null i.e. [null, null]. I'm able to get other information like server status, timestamp etc.

I tried it using node-red-contrib-iiot-opcua & node-red-contrib-opcua.

I saw a similar issue opened in node-red-contrib-iiot-opcua but it is not working with normal opc ua contribs also. Is there an alternative solution to this for what can be done in order to retrieve this data? Thanks in advance.

mikakaraila, 2019-10-09:
I expect node is custom, but can you attach screenshot from UaExpert that shows structure?

mikakaraila, 2019-10-09:
Current version supports only basic types.

piyush-gupt, 2019-10-11:
Here is the screenshot of UA export. It shows an array with two structure elements. I'm only getting null in array values in node red
[image]

mikakaraila, 2019-10-14:
Hmm, it should be loaded somehow into the code. At the moment this is not possible.
There should be a way to expand types and load them into the runtime. 
Perhaps extra node that could do it, no time to implement now.

piyush-gupt, 2019-10-14:
Can you suggest how can I load the data types in the runtime? Is there a place where I can define them and then give its reference to the actual node?

mikakaraila, 2019-10-14:
One way would be to add them as own separate XML. This could be loaded with own namespace and added as own types. Then those could be used as I have done earlier, but not with node-red.

mikakaraila, 2021-06-10:
I have done some progress with this, code is not yet complete and example use missing.

mikakaraila, 2023-01-24:
Please test with new v0.2.295 as it uses now node-opcua v2.89 that contains fixes to extension objects.

mikakaraila, 2023-07-04:
No activity or test result provided => closed. Reopen with new test results.

closed 2023-07-04 by mikakaraila