mikakaraila/node-red-contrib-opcua #123: OPC-UA Server restart -> TransferSubscriptionRequest

bits-bytes, 2019-11-03:
Hello,
I encounter a problem while reconnection the OPC Ua client (node-red) after my OPC-UA Server is restarting (the PLC hosting the Server is restarted, so is the OPC UA server as well).
In the reconnection process it comes to a point where a new session is created. Then the client (OPC UA Client is requesting a TransferSubscriptionRequest. I have attached a picture, I hope it shows.
My OPC Ua Server has not implemented this service, anyhow in this situation it could not transfer nothing as because of the reboot, noting is available.
In reconnection.js then is a comment that reconnecting does not work if the server crashes unexpectedly ....

How is your idea about this issue, is it planned to resolve it in the future?

Thank you
BR Manfred

[image]

00:00:27.804Z :client_session_keepalive_manag:53    ClientSessionKeepAliveManager#ping_server  49887 60000
make_debug_log.js:57
 warning : ClientSessionKeepAliveManager#ping_server  Connection Break
client_session_keepalive_manager.js:57
00:00:27.819Z :reconnection                  :294    Starting sessions reactivation
make_debug_log.js:57
00:00:27.823Z :reconnection                  :272     TRYING TO REACTIVATE EXISTING SESSION  ns=1;i=2233142146
make_debug_log.js:57
00:00:27.825Z :reconnection                  :273        SubscriptionIds : [ 2233142144 ]
make_debug_log.js:57
00:00:27.833Z :reconnection                  :280       ActivateSession :   ServiceResult is BadSessionIdInvalid (0x80250000) request was ActivateSessionRequest
make_debug_log.js:57
00:00:27.837Z :reconnection                  :117    repairing client session by_recreating a new session  ns=1;i=2233142146
make_debug_log.js:57
00:00:27.840Z :reconnection                  :134       => suspend old session publish engine....
make_debug_log.js:57
00:00:27.843Z :reconnection                  :143       => creating a new session ....
make_debug_log.js:57
00:00:27.856Z :opcua_client                  :265   revised session timeout =  60000
make_debug_log.js:57
00:00:27.859Z :reconnection                  :147       => creating a new session (based on old session data).... Done
make_debug_log.js:57
00:00:27.862Z :reconnection                  :160       => activating a new session ....
make_debug_log.js:57
00:00:27.867Z :reconnection                  :163       =>  activating a new session .... Done
make_debug_log.js:57
00:00:27.871Z :reconnection                  :176     session subscriptionCount =  1
make_debug_log.js:57
00:00:27.874Z :reconnection                  :181       => asking server to transfer subscriptions = [ 2233142144 ]
make_debug_log.js:57
00:00:27.976Z :client_session                :821   ClientSession#emitCloseEvent

mikakaraila, 2019-11-03:
Yes, I and Etienne we both know this problem. It is depending how server is implemented.
Etienne fixed this to node-opcua v2.2 but it is TypeScript based.
I should re-write nodes with TypeScript and then compiled result could be used.
No time at the moment to start this work.

bits-bytes, 2019-11-03:
Hi Mika, thank you for your fast answer.
I understand this is not on the todo list for you right now. 
Maybe you have an **idea** about a temporarily workaround for the problem, that you would want to share?
If I could figure out how to simply restart the hole client (like restart of the flows) in this case. I was lurking around the code but I am not getting firm with it :-).
Anyhow, I will see and wait how it continues.
Again thanks for your time.
BR Manfred

mikakaraila, 2019-11-03:
Check how Etienne made it:
https://github.com/node-opcua/node-opcua/issues/678

bits-bytes, 2019-11-03:
Ok,
I will try my luck and investigate it... :-)
Thanks again.

mikakaraila, 2020-07-07:
node-opcua re-connects automatically and if server does not support TransferSubscriptionRequest then code creates new subscription and clones items to it.

closed 2020-07-07 by mikakaraila

bits-bytes, 2020-07-08:
Hi Mika,
thank you for this.

I already tested it, but I have to say I have exactly the same debug-output and behavior like before.
See screenshot at issue  #123

Node red indicates that I have the right version in use (2 days old).

I was not working for some time now on this project, but now I see that OPC-UA IIOT is depreciated by now.
Can this be the reason?
Should I change over OPC-UA Item and so on?

Sorry to bother you with this, I can see my problem is I don’t see the big picture of what components work together.
I already could help myself if you only could point me to an information document about the connection of
OPC-UA, IIOT packages and so on.

[cid:image002.jpg@01D6554B.612A2B30]

These nodes I use now....
From the package OPCUA-IIOT (bianco)

[cid:image004.jpg@01D6554B.612A2B30]

I have around 400 OPC Ua Items

I think I also could use this one (OPCUA Client/Item), although than I need 400 of them?
[cid:image006.jpg@01D6554B.612A2B30]
Thank you very much and I hope you can still send me some information so I can continue on the correct path

BR
Manfred