mikakaraila/node-red-contrib-opcua #126: Can't be installed on Ubuntu

mstfldmr, 2019-11-12:
OS: Ubuntu 18.04.3
Nodejs: v12.13.0
Node-RED: v1.0.2

```

# npm install node-red-contrib-opcua

> node-opcua-client@0.7.4 postinstall /usr/lib/node_modules/node-red/node_modules/node-opcua-client
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem

Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/node-red/node_modules/node-opcua-client/certificates'
    at Object.mkdirSync (fs.js:823:3)
    at Object.mkdir (/usr/lib/node_modules/node-red/node_modules/node-opcua-pki/lib/pki/toolbox.ts:79:12)
    at /usr/lib/node_modules/node-red/node_modules/node-opcua-pki/lib/crypto_create_CA.ts:211:9
    at cb (util.js:196:39)
    at processTicksAndRejections (internal/process/task_queues.js:80:21)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-client@0.7.4 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-opcua-client@0.7.4 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-12T18_33_11_651Z-debug.log
```

mikakaraila, 2019-11-12:
Did use sudo or root as you tried to install?

mstfldmr, 2019-11-12:
I tried doing it as root

mikakaraila, 2019-11-13:
I suppose postinstall is reported on node-opcua #699

mikakaraila, 2019-11-17:
Closed as node-opcua is actually causing installation problem.

closed 2019-11-17 by mikakaraila