mikakaraila/node-red-contrib-opcua #131: NodeRed is crashing when OPCUA node disconnected from debug node

akvish, 2020-01-06:
I was running node-red with 100 item nodes connected to one client node with read option. It was running perfectly fine and able to see the output in debug panel. But when I disconnect debug node and redeploy, node-red is crashing with following error message

 **TypeError: Cannot read property 'getEndpoints' of null
    at /root/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:254:27
    at /root/.node-red/node_modules/async/dist/async.js:2952:28
    at replenish (/root/.node-red/node_modules/async/dist/async.js:444:21)
    at iterateeCallback (/root/.node-red/node_modules/async/dist/async.js:428:21)
    at /root/.node-red/node_modules/async/dist/async.js:325:20
    at /root/.node-red/node_modules/async/dist/async.js:2957:17
    at Timeout.delayer [as _onTimeout] (/root/.node-red/node_modules/delayed/delayed.js:27:10)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-red@1.0.3 start: `node packages/node_modules/node-red/red.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-red@1.0.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-06T06_13_18_347Z-debug.log**

mikakaraila, 2020-01-07:
Did you deploy whole flow or just modified? Can you run with verbose logging?
node-red -v TEST_FLOW.json
Attach console output, please.

mikakaraila, 2020-07-07:
I expect this as solved with new version

closed 2020-07-07 by mikakaraila