mikakaraila/node-red-contrib-opcua #133: Can't write Data in PLC

Jelazzaoui, 2020-02-11:
Hallo maikakraila, I am using node-red-contrib-opcua client to write and read Data from PLC S7-1512. Reading data works with no problems but when i try to write in the PLC i receive this errors:
Error at assert (/home/pi/.node-red/node_modules/node-opcua-assert/dist/index.js:6:21) at new NodeId (/home/pi/.node-red/node_modules/node-opcua-nodeid/src/nodeid.js:86:5) at write_action_input (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:722:18) at OpcUaClientNode.processInputMsg [as _inputCallback] (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:429:11) at OpcUaClientNode.Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:199:18) at Immediate._onImmediate (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:179:33) at processImmediate (internal/timers.js:439:21)

I wish you could help me to solve this issue.
Best regards
Jalal
[image]

mikakaraila, 2020-02-11:
Please check first with UaExpert that nodeId is correct.
Is namespace ns=3 ? Or some other index number?

mikakaraila, 2020-02-11:
What is in your flow JSON / string that you inject?
If you use browseName it requires namespace index this way:
 // Parse nodeId string, s=1:PST-007-Alarm-Level@Training?SETPOINT

Jelazzaoui, 2020-02-11:
I use OPC Scout V10 to get the the namespace, see the screenshot below:
[image]
Thats what i inject: NodeID;ns=3;s="Baustein_1_DB"."zahl";datatype=Float

mikakaraila, 2020-02-11:
It could be something with " characters, just wild idea.
Try:
ns=3;s=\"Baustein_1_DB\".\"zahl\";datatype=Float

Jelazzaoui, 2020-02-11:
I tried that, now i dont get errors but i still dont write anything in my PLC

mikakaraila, 2020-02-11:
What comes to console?
run with "node-red -v FLOWFILE.json

mikakaraila, 2020-02-11:
Do you inject nodeId as json?
Can you change it to item node?

[image]

Jelazzaoui, 2020-02-11:
Cosole:
`11 Feb 14:51:47 - [info] Flows starten
11 Feb 14:51:47 - [info] Flows gestartet
11 Feb 14:51:47 - [info] Server wird jetzt auf http://127.0.0.1:1880/ ausgeführt.
11 Feb 14:55:16 - [info] Flows stoppen
11 Feb 14:55:16 - [info] Flows gestoppt
11 Feb 14:55:16 - [info] Flows starten
11 Feb 14:55:16 - [info] [OpcUa-Client:Write]   No certificate used.
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [warn] [OpcUa-Client:Write] Write: create Client ...
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Client status: create client
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] async series - connecting
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Client status: connecting
11 Feb 14:55:16 - [info] [OpcUa-Client:Write]   No certificate used.
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [warn] [OpcUa-Client:Write] Write: create Client ...
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] [object Object]
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Client status: create client
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] async series - connecting
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Client status: connecting
11 Feb 14:55:16 - [info] Flows gestartet
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] endpoint opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Product URI https://www.siemens.com/s7-1500
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application Name SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Security Mode NONE
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#None
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Type SERVER
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] certificate ... /*endpoint.serverCertificate*/
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] discoveryUrls opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] endpoint opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Product URI https://www.siemens.com/s7-1500
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application Name SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Security Mode SIGN
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Type SERVER
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] certificate ... /*endpoint.serverCertificate*/
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] discoveryUrls opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] endpoint opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Product URI https://www.siemens.com/s7-1500
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application Name SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Security Mode SIGNANDENCRYPT
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Type SERVER
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] certificate ... /*endpoint.serverCertificate*/
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] discoveryUrls opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] endpoint opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Product URI https://www.siemens.com/s7-1500
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application Name SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Security Mode SIGN
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Type SERVER
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] certificate ... /*endpoint.serverCertificate*/
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] discoveryUrls opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] endpoint opc.tcp://192.168.0.1:4840
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Product URI https://www.siemens.com/s7-1500
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Application Name SIMATIC.S7-1500.OPC-UA.Application:PLC_1
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Security Mode SIGNANDENCRYPT
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] Type SERVER
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] certificate ... /*endpoint.serverCertificate*/
11 Feb 14:55:16 - [info] [OpcUa-Client:Write] discoveryUrls opc.tcp://192.168.0.1:4840
[... cut ...]
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] namespace=3
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] string=\"Baustein_1_DB\".\"zahl\"
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] type=Float
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] value=1581429331369
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] ns=3;s=\"Baustein_1_DB\".\"zahl\"
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] Client status: active writing
11 Feb 14:55:31 - [info] [OpcUa-Client:Write] Value written!
 Server State =  Running
 Server State =  Running`

mikakaraila, 2020-02-11:
Client says: 
"Client status: active writing 11 Feb 14:55:17 - [info] [OpcUa-Client:Write] Value written!"

Next question is there other program writing to same DB? At PLC program perhaps?

Jelazzaoui, 2020-02-11:
It works fine now! I just removed the slaches \\ . I have no idea why is it working right now. But anyway thank you so much for your help!

mikakaraila, 2020-02-11:
Good, most probably something was wrong in the original flow

closed 2020-02-12 by mikakaraila