mikakaraila/node-red-contrib-opcua #138: Node Red keep crashing after many communication errors

roupec, 2020-03-02:
[00 - notes.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4276644/00.-.notes.txt)
ERROR !!! , please check here !!!! callback may be called twice !!  Error: Callback was already called.
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:318:36
    at _iteratee (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:245:17)
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_base.js:885:13
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:752:9
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:717:13
    at Object.<anonymous> (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_session.js:1169:20)
    at Object.modified_callback [as callback] (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1015:28)
    at ClientSecureChannelLayer.cancelPendingTransactions (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1308:25)
    at ClientSecureChannelLayer.close (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1329:10)
    at OPCUAClientBase.disconnect (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_base.js:981:21)
ERROR !!! , please check here !!!! callback may be called twice !!  Error: Callback was already called.
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:318:36
    at _iteratee (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:245:17)
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_base.js:885:13
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:752:9
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:717:13
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_session.js:1169:20
    at Object.modified_callback [as callback] (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1019:13)
    at ClientSecureChannelLayer.cancelPendingTransactions (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1308:25)
    at ClientSecureChannelLayer.close (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1329:10)
    at OPCUAClientBase.disconnect (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_base.js:981:21)
2 Mar 23:56:21 - [red] Uncaught Exception:
2 Mar 23:56:21 - Error: Callback was already called.
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:318:36
    at _iteratee (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/node_modules/async/dist/async.js:245:17)
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_base.js:885:13
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:752:9
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/opcua_client.js:717:13
    at /home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-client/src/client_session.js:1169:20
    at modified_callback (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1019:13)
    at Timeout._onTimeout (/home/pedronel/.node-red-EPMSmain/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1030:9)
    at listOnTimeout (internal/timers.js:535:17)
    at processTimers (internal/timers.js:479:7)

mikakaraila, 2020-03-05:
What is the first error?
Are computers on same time?

**2 Mar 23:56:20** - [error] [OpcUa-Client:reliance-itz-100] reliance-itz-100 Cannot write value (257) to msg.topic:ns=2;s=Server/Server/GMIS/GMIS_TKPS_TIMER error:Error: Transaction has timed out ( timeout = 60000 ms)
 Timeout .... waiting for response for  WriteRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;b=4a2e0594a93edd5f184fdda1e76e1e4424f4aa1721e078a4751744bd8045a858
 timeStamp                     /* UtcTime                          */: **2020-03-02T15:55:20.442Z**

roupec, 2020-03-08:
Hello,

Thank you for the tip in regards of the time. I have find out that OPC UA server we are connecting to is lagging time. Time on OPC UA server is lagging about 70 seconds per hour.

Let me verify.

Thank you,

Petr

roupec, 2020-03-08:
[LOP1458 - Crashing Node Red.numbers-Sheet 1.pdf](https://github.com/mikakaraila/node-red-contrib-opcua/files/4303395/LOP1458.-.Crashing.Node.Red.numbers-Sheet.1.pdf)

mikakaraila, 2020-03-08:
Etienne is working with new v2.5 there will be some fixes related to computer clock jump.

mikakaraila, 2020-07-07:
I expect this as solved as no comments lately

closed 2020-07-07 by mikakaraila