mikakaraila/node-red-contrib-opcua #139: Raspberry Pi

mhechthz, 2020-03-09:
Hi, 

I tried to install this node on raspberry pi but failed.

Is it possible to use this node on raspberry pi?

mikakaraila, 2020-03-09:
Should work, check --safe_perm option it is needed for node-opcua

mhechthz, 2020-03-10:
`pi@raspberrypi:~ $ sudo npm install -g node-red-contrib-opcua --safe_perm

> deasync@0.1.19 install /usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync
> node ./build.js

gyp WARN EACCES current user ("nobody") does not have permission to access the dev dir "/root/.cache/node-gyp/12.16.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync
gyp ERR! node -v v12.16.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
Build failed
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! deasync@0.1.19 install: `node ./build.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the deasync@0.1.19 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-10T07_13_52_049Z-debug.log
pi@raspberrypi:~ $ 
`

mikakaraila, 2020-03-10:
Or the option is --unsafe-perm

mhechthz, 2020-03-10:
```
pi@raspberrypi:~ $ sudo npm install -g node-red-contrib-opcua --unsafe_perm

> deasync@0.1.19 install /usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync
> node ./build.js

gyp WARN EACCES current user ("nobody") does not have permission to access the dev dir "/root/.cache/node-gyp/12.16.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/node-red-contrib-opcua/node_modules/deasync
gyp ERR! node -v v12.16.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
Build failed
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! deasync@0.1.19 install: `node ./build.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the deasync@0.1.19 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-10T08_22_01_565Z-debug.log
pi@raspberrypi:~ $ 

```

nospam2000, 2020-03-10:
There is still a typo: `--unsafe_perm` => `--unsafe-perm`. See also [npm config docs](https://docs.npmjs.com/misc/config).

I would recommend not to install any packages globally, that means I would `cd` to the project directory and install it there without the `-g` switch:
```
cd mynodproject
npm install node-red-contrib-opcua --save-prod
```

That makes it possible to have different versions for different projects, which I need very often. Otherwise it could break your working project when you install or update the global packages because of a completely unrelated test project.

mhechthz, 2020-03-10:
Perfect ... this did the job. Thank You!

closed 2020-03-10 by mhechthz