mikakaraila/node-red-contrib-opcua #140: How to set samplingInterval

mhechthz, 2020-03-20:
Hi,

I try to register for a lot of variables that I get from database. I prepare the data as follows:

```
var data = msg.payload;

arr = [];

for(i=0;i<data.length;i++)
{
    try{
        entry = {
               'topic': "ns=2;s=HO5OPC.Forschung."+data[i].tagname,
               'datatype':data[i].datatype,
               'payload':Date.now(),
               'name':data[i].name,
               'samplingInterval':data[i].scanrate+"ms",
                };
       arr.push(entry);
    }
    catch(e)
    {
       
    }
};

msg = arr;

return [msg];
```
This might not be the most elegant way, but works (I'm rather new to Node-Red).
What doesnt work is the samplingInterval. I tries with and without "ms", but all variables come "as often as they want or as they change".

How can I configure, that the varables come only in defined intervals, e.g. 0.5, 2, 5 or 10 seconds?

mikakaraila, 2020-03-22:
You have to use msg.payload directly then it is used as a parameter for making monitored item to subscription.

mhechthz, 2020-03-22:
Hmmm ... the shown approach works well for subscription of multiple variables, except for samplingInterval. 

If I go back to msg.payload, than I can only subscribe for ** one ** variable. I need to subscribe for >> 500 variables which I will not configure by hand. What I would need is to hand over the samplingInterval to opcua-client node instead of configuring the samplingInterval manually inside the opcua-client node.

mikakaraila, 2020-09-02:
msg.interval and msg.queueSize are now available

closed 2020-09-02 by mikakaraila