mikakaraila/node-red-contrib-opcua #141: How to set deadbandType and deadbandValue

mhechthz, 2020-03-21:
I didn't find it in the documentation, but it seems, that the default behaviour of SUBSCRIBE is an absolute deadband with an deadbandValue of 1.

How can this be changed?

I would like to send something like this to the node:
```
{
               'topic': "ns=2;s=HO5OPC.Forschung."+data[i].tagname,
               'datatype':h[data[i].datatype],
               'payload':Date.now(),
               'name':data[i].name,
                'samplingInterval':1000,
                deadbandType: 'PercentDeadband',
                deadbandValue: 0.1
}
```
but it is not clear what is the right sysntax or if it works like this at all.

mikakaraila, 2020-03-22:
Just read code:

var interval = convertAndCheckInterval(msg.payload);
verbose_log(msg.topic + " samplingInterval " + interval);
verbose_warn("Monitoring: " + msg.topic + ' by interval of ' + interval + " ms");
verbose_log("Deadband type (a==absolute, p==percent) " + node.deadbandtype + " deadband value " + node.deadbandvalue);

mhechthz, 2020-03-22:
Thank you. I will try this.

mhechthz, 2020-08-25:
Added this to my node, but it didn't work:
```

subs = {
       'topic': "ns=3;s=Sinusoid",
       'datatype':"Float",
       'payload': 500, // subscription frequency
       'name': "Sinusoid",
       'deadbandtype':'a',
       'deadbandvalue':0.2
}

return subs;
```

mikakaraila, 2020-09-02:
Sorry I forgot it is inside node, not in msg object.

mhechthz, 2020-09-09:
Ok, meanwhile I did some testing with `deadbandtype` and `deadbandvalue`.

Both cannot be set via nodes, so I added it via msg that is sent to OPC-UA Client node (as described above):
```
subs = {
 'topic': "ns=3;s=Sinusoid",
 "action": "subscribe",
 'datatype':"Float",
 'interval': 200, // subscription frequency
 'deadbandtype':'p',
 'deadbandvalue':30
}

return subs;
```

This has no influence on the behaviour.

I found that both `deadbandtype` and `deadbandvalue` are hard coded in the OPC-UA Client node. So I did an export of the node, changed the values and did a re-import:

  ```
  {
        "id": "fc8ead94.9314c",
        "type": "OpcUa-Client",
        "z": "6e71ea6f.c1a4d4",
        "endpoint": "a2742222.82122",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 2,
        "time": "5000",
        "timeUnit": "ms",
        "certificate": "n",
        "localfile": "",
        "name": "",
        "x": 560,
        "y": 440,
        "wires": [
            [
                "18d64a3.01aedb6"
            ]
        ]
    }
```
This also has no influence on the behaviour. So, what to change in the code?

mikakaraila, 2020-09-09:
Action must be MONITOR, subscription does not use those settings.

[image]

mhechthz, 2020-09-09:
Ok. I try this. Thank you.

mikakaraila, 2020-09-09:
I have to add note to new readme about this

mhechthz, 2020-09-10:
Hmmm ... seems that the deadband can only be configured via the node but not via the message object. This means, that for hundreds of different variables I cannot use a configuration database. Do you agree? Do you plan to change this.

mikakaraila, 2020-09-10:
I noticed this already, needs just to handle input message. I will make enhancement to next version.

mikakaraila, 2020-09-10:
You can use now:

msg.deadbandType == "a" or msg.deadbandType == "p"

and

msg.deadbandValue

closed 2020-09-11 by mikakaraila