mikakaraila/node-red-contrib-opcua #142: OPCUA-Client crashes

mhechthz, 2020-03-21:
Got the following while playing with OPCUA connection:
```
Mar 09:16:33 - [error] [OpcUa-Client:2bab8478.7c27fc] Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2020-03-21T09:16:33.912Z
   requestHandle               /* IntegerId                        */: 103               0x67
   serviceResult               /* StatusCode                       */: BadSessionClosed (0x80260000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
21 Mar 09:16:34 - [red] Uncaught Exception:
21 Mar 09:16:34 - TypeError: Cannot read property 'getEndpoints' of null
    at /usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:254:27
    at eachfn (/usr/src/node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:2948:28)
    at replenish (/usr/src/node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:440:21)
    at iterateeCallback (/usr/src/node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:424:21)
    at /usr/src/node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:321:20
    at result (/usr/src/node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:2953:17)
    at Timeout.delayer [as _onTimeout] (/usr/src/node-red/node_modules/delayed/delayed.js:27:10)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-red-docker@1.0.4 start: `node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-red-docker@1.0.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-03-21T09_16_34_848Z-debug.log

```

mikakaraila, 2020-03-22:
Server does not provide endpoint, are url and port correct ones?

mhechthz, 2020-03-22:
Well, at this point I thought yes. I will keep a eye on the behaviour to see, if it is reproducible.

I hope I don't annoy you, but I try to apply the node-red-contrib-opcua in a challenging context to read about 600 variables from a KEPserver for a research project.

Currently we still struggle to configure the server and node-red to work properly together. UncoughtExteption terrifies me a little bit.

mikakaraila, 2020-03-23:
OK I expect you read configuration from file and fetch it then through function node to opcua client node. We have some experience KepServer on customer installations.

mhechthz, 2020-03-23:
> OK I expect you read configuration from file and fetch it then through function node to opcua client node.
YES.

I'm really in a mess. If I work with node-red-contrib-iiot-opcua and the setting "values only" I can subscribe and get back all values as expected, but as they stuck ion Node-OPCUA 0.7.x the topic is not sent back, so I cannot use this approach. But it shows, that **the KEPserver is configured properly**. 

If I de-select "values only", the node-red-contrib-iiot-opcua crashes the node-red-server since too much data are sent.

If I use node-red-contrib-opcua and READ values from KEPserver, all works well If I than SUBSCRIBE, I get not back any values but those at subscription. The exactly same aproach works with Prosys demo server. So there seems to be a certain issue with KEPserver and node-red-contrib-opcua for SUBCRIBTION.

mhechthz, 2020-03-23:
Is there any documentation for node-red-contrib-opcua, which properties can be injected to opcua-client-node?

      ```
  entry = {
               'topic': "ns=2;s=varname',
               'datatype':'Float',
               'payload':Date.now(),
               'name':'readable name',
               'requestedPublishingInterval':2000,
               'samplingInterval': 2000,
               'deadbandType': 'p',
               'deadbandValue': 0.1,
               'injectType': 'subscribe',
               'publishingEnabled': true,
                };
```
I don't see much effect of the  values starting from requestedPublishingInterval.
Where can I find the properties that will be taken into acount?

mikakaraila, 2020-03-23:
Most on the properties are fixed, look source code 102-opcuaclient.js

mhechthz, 2020-03-23:
:+1: :1st_place_medal: unbelievable. Checking the sourcecode, I see, that the **msg.payload** is mapped on interval and neither the 'requestedPublishingInterval' nor the 'samplingInterval'. I'm not sure wether this is a bug or a feature. Since I sent the current timestamp via msg.payload, I have to wait billions of seconds to get an update ...

So finally my request would be to accept the free parameters as e.g. noted above, e.g. via msg.samplingInterval.

mikakaraila, 2020-03-23:
OK are all 600 variables now subscribed?

mhechthz, 2020-03-23:
Yes, all 617 variables subscribed and comming stable every "msg.payload" ms. Since I get the configuration from database, the flow is now quite compact reading all configured variables from postrgres, injecting for subscribtion to opcua-client and store the results back to a postgres/timescaldb (with some corrections on timestamp).

Thank you very much.

closed 2020-04-22 by mikakaraila