mikakaraila/node-red-contrib-opcua #146: Subscribe or Monitor on opcUACLient node doesn't update value

henjoe, 2020-05-07:
I have a simple flows of node. 
Inject (with topic of nodeID I want to subscribed) -> OPCUAClient node (With subscribe /Monitor as Action), I set the interval as 1 seconds. 

I assumed that this node will output the value of specific OPC items every seconds? Am i right on this? But when I press the inject node, the opcuaClient node just changed to "KeepAlive" status, and there were no output. 
Here is my configuration on my client node:
[image]

here is the flow:
[image]

Debug has only 1 ouput, and it never continues, what I want is every seconds, what ever the value of the OPC, it will give me the output.
[image]

henjoe, 2020-05-08:
I have another observation regarding this issue, the "Subscribe" function seems to be working when OPCUA Server is in local machine. But when the UA server is on another machine, the subscribe is not working well. 

I don't know if this is normal or bug.
Note also that Read is working even on a remote UA Server, I just don't know why it didnt work on subscribe, the node stock on "Keep Alive" status. 

[image]

mikakaraila, 2020-05-08:
Use SUBSCRIBE I just tested and it is working:
[image]

mikakaraila, 2020-05-08:
Hmm, are you sure that connection is created? Can you activate -v verbose when you start your flow?
node-red -v flow.json
CHeck also server side that session is active and test connection with UaExpert and look server diagnostics -> node-red client -> session information

henjoe, 2020-05-08:
Hello Thank you for your prompt answer. And yes! I can access the remote UA server using the OPCUA Expert, that is why I am wondering why the subscribrle not working on that UA server (remote).

Have you already tried accessing a remote UA server ? If it works to you, maybe I just need more configuration on the side of UA server. 

Again, thank you.

henjoe, 2020-05-08:
This is what I have on my command line after running Node-red -v, And when I start my flow

[image]

mikakaraila, 2020-05-08:
Is value changing on server side? Subscription will update value to client only if it changes

henjoe, 2020-05-08:
Ah.. see, Okay I got it. it seems that the OPC tag should be updating. Everytime it updates then it pushes the value. 
Is there a way that whether the value changes or not, it pushes the data of OPC tags?  
I do believe I can do this using the READ, but what I like to subscribe is it gives information of the SourceTimestamp which is very important on my side. Is there a way to do this? 

Again, thank you.

mikakaraila, 2020-05-09:
I change single read to other function that will return also timestamps

closed 2020-05-09 by mikakaraila