mikakaraila/node-red-contrib-opcua #147: Internal error on READ operation

pfetrial, 2020-05-08:
hello
If I read an byte item in a OPC UA server, I obtain an"internal error" . Write Operation is OK. 
Log is:
Welcome to Node-RED
===================
 
7 May 08:53:17 - [info] Node-RED version: v1.0.4
7 May 08:53:17 - [info] Node.js  version: v12.16.3
7 May 08:53:17 - [info] Windows_NT 10.0.17763 x64 LE
7 May 08:53:18 - [info] Loading palette nodes
7 May 08:53:20 - [info] Dashboard version 2.21.0 started at /ui
7 May 08:53:20 - [info] Settings file  : \Users\i00204714\.node-red\settings.js
7 May 08:53:20 - [info] Context store  : 'default' [module=memory]
7 May 08:53:20 - [info] User directory : \Users\i00204714\.node-red
7 May 08:53:20 - [warn] Projects disabled : editorTheme.projects.enabled=false
7 May 08:53:20 - [info] Flows file     : \Users\i00204714\.node-red\flows_CI-13574.json
7 May 08:53:20 - [info] Server now running at http://127.0.0.1:1880/
7 May 08:53:20 - [warn]
 
---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.
 
If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.
 
You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------
 
7 May 08:53:20 - [info] Starting flows
7 May 08:53:20 - [info] [OpcUa-Client:f5308d04.6467d]   No certificate used.
7 May 08:53:20 - [info] [OpcUa-Client:ce3f8a9e.4d4d38]  No certificate used.
7 May 08:53:20 - [info] Started flows
ERROR !!! callback has thrown en error  TypeError: Cannot read property 'bold' of undefined
    at C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:487:58
    at C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\dist\private\client_session_impl.js:340:13
    at C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\dist\private\client_session_impl.js:184:17
    at ClientSecureChannelLayer.<anonymous> (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\dist\private\client_session_impl.js:779:20)
    at modified_callback (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:892:31)
    at process_request_callback (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:65:5)
    at ClientSecureChannelLayer._on_message_received (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:454:9)
    at MessageBuilder.<anonymous> (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:139:18)
    at MessageBuilder.emit (events.js:310:20)
    at MessageBuilder._decodeMessageBody (C:\Users\i00204714\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\message_builder.js:190:26)
7 May 08:53:51 - [error] [OpcUa-Client:ce3f8a9e.4d4d38] Error: internal Error
 
What is the problem ?
Thanks

mikakaraila, 2020-05-08:
I will check tomorrow, most probably missing require for chalk or something similar

mikakaraila, 2020-05-09:
Fixed, need to add colors and fixed dataType.toString() at the same time.
Code needed some updates.

closed 2020-05-09 by mikakaraila

pfetrial, 2020-05-09:
thanks a lot . I will try

pfetrial, 2020-05-10:
Hello

Every thing works fine thanks.

pfetrial