mikakaraila/node-red-contrib-opcua #148: Error: get-installed-path: module not found "node-opcua-server" in path

hilschernetpi, 2020-05-11:
This issue relates back to issue https://github.com/mikakaraila/node-red-contrib-opcua/issues/110 and I am getting the same error with latest opc-ua node in the version 0.2.52.

To reconstruct the problem execute the following commands to install node.js, npm and Node-RED and then finally the opcua nodes

```
curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -
apt-get install -y nodejs
npm install -g --unsafe-perm node-red@1.0.6
cd /usr/lib
npm install node-red-contrib-opcua@0.2.52
```
What you then get is the following folder structure

/usr/lib/node_modules/
   node-red-contrib-opcua
      |
      |- node_modules
      |        |
      |        | - node-opcua-server
      |        | ....
      |
      | - opcua
      |      |
      |      | - 104-opcuaserver.js
      |      | ....
      |
npm
  |
node-red

If I now start node-red and want to place the opc-ua-server node, I get the error Error: get-installed-path: module not found "node-opcua-server" in path

I searched for the problem and I examined the the source code @0.2.52 and in detail the search paths again in 104-opcuaserver.js like it was done in issue https://github.com/mikakaraila/node-red-contrib-opcua/issues/110 

I see the current version
```
var serverPkg = installedPath.getInstalledPathSync('node-opcua-server', {
    paths: [
         path.join(__dirname, '..'),
        path.join(__dirname, '../..'),
         path.join(process.cwd(), './node_modules'),
         path.join(process.cwd(), '.node-red/node_modules'),
     ],
})
```

For my installation constellation in the list of search paeth there should be an additional path.

```
var serverPkg = installedPath.getInstalledPathSync('node-opcua-server', {
    paths: [
         path.join(__dirname, '..'),
         path.join(__dirname, '../..'),
         path.join(process.cwd(), './node_modules'),
         path.join(process.cwd(), '../node_modules'),
         path.join(process.cwd(), '.node-red/node_modules'),
     ],
})
```
I edited it in my version for a test and indeed the problem dissappeared. 

Thank you
Armin

mikakaraila, 2020-05-11:
OK added and pushed to git, but I will wait if there will come other fixes.
I just published new versions.

mikakaraila, 2020-05-11:
v0.2.57 contains this now

closed 2020-05-11 by mikakaraila