mikakaraila/node-red-contrib-opcua #149: automatically restart/reconnect client node on runtime

mhechthz, 2020-05-14:
Hi,

for the last weeks my opcua-client with over 600 subscribed variables and about 5000 messages per minute did run quite well. Rare crashes of opcua node I can usually catch by automatically restarting the node-red-docker-container (which isn't nice but works).

From time to time it happens that there arrive no more messages even if the client node shows "green" on connection and the opcua-server is running (probably network problems). If I restart the workflow manually (by deploy) all runs again well. So I added a watchdog node that can react on message timeouts.

Is there any chance to automatically restart/reconnect the client node on runtime? Only inject the variabes for subscription again doesn't help.

mikakaraila, 2020-05-14:
I found one low level error from node-opcua today. Write will crash something in some rare cases. Waiting to Etienne check code... but latest v2.5.12 is otherwise good.
Session re-connect problem somehow. 
It is inside node.opcua that will negotiate connection back alive.

mikakaraila, 2020-05-14:
I updated latest node-red-contrib-opcua v0.2.57 to flows and npm.

mhechthz, 2020-05-14:
Sounds promising. I really like doing (semi-)professional opc-ua connection in node-red for industrial application (steel industry). 

So waiting for your go for updating the node.

mikakaraila, 2020-05-15:
One good solution is to use other flow to check that server current time is changing.
If it is not changing then you should re-start OPCUA flow. Node-red REST API can be used to do trick.

mhechthz, 2020-05-17:
Well I tried restarting with Node-red REST API but get permission denied. But this has nothing to do with OPC-UA.

mikakaraila, 2020-05-18:
https://nodered.org/docs/api/admin/oauth

mhechthz, 2020-05-18:
Its running now. Thank you for your kind assistance. I will observe the behaviour of the node and keep you informed if anything strange happens.

closed 2020-08-05 by mikakaraila