mikakaraila/node-red-contrib-opcua #151: "node-red-contrib-iiot-opcua" IS OK,  "node-red-contrib-opcua" has errors on same tags!  PLEASE HELP

youngkzy, 2020-05-19:
I used the older "node-red-contrib-iiot-opcua" nodes and currently am using these with success.  However, they are deprecated and no longer available for install.

 I'm trying to move to the "node-red-contrib-opcua" nodes and I get errors trying to connect to any of my current Kepware OPC Servers.  And yes, they work fine in local and remote with the other older nodes.  I do not know what to do in order to troubleshoot this. Can someone PLEASE HELP !!

Here is a copy from the console.
There is nothing defined in my nodes or configuration with 
"ns=0;i=449"

18 May 17:49:36 - [warn] [OpcUa-Browser:6363eeea.6e0a4] input browser
18 May 17:49:36 - [info] [OpcUa-Browser:6363eeea.6e0a4] start browse client on opc.tcp://127.0.0.1:49320
21:49:36.796Z :message_builder_base          :153   Error   client68 cannot construct object with nodeID ns=0;i=449
18 May 17:49:42 - [error] [OpcUa-Client:82f9e500.e28b98]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:49:42 - [error] [OpcUa-Browser:6363eeea.6e0a4] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:49:42 - [info] [OpcUa-Browser:6363eeea.6e0a4] Browse loading Items done ...
18 May 17:49:45 - [error] [OpcUa-Browser:6363eeea.6e0a4] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:49:45 - [info] [OpcUa-Browser:6363eeea.6e0a4] Browse loading Items done ...
18 May 17:50:19 - [info] Stopping modified nodes
18 May 17:50:19 - [info] Stopped modified nodes
18 May 17:50:19 - [info] Starting modified nodes
18 May 17:50:19 - [info] [OpcUa-Browser:6363eeea.6e0a4] start browse client on opc.tcp://127.0.0.1:49320
18 May 17:50:19 - [info] Started modified nodes
21:50:19.810Z :message_builder_base          :153   Error   client69 cannot construct object with nodeID ns=0;i=449
18 May 17:50:24 - [warn] [OpcUa-Browser:6363eeea.6e0a4] input browser
18 May 17:50:24 - [info] [OpcUa-Browser:6363eeea.6e0a4] start browse client on opc.tcp://127.0.0.1:49320
21:50:24.760Z :message_builder_base          :153   Error   client70 cannot construct object with nodeID ns=0;i=449
18 May 17:50:31 - [error] [OpcUa-Browser:6363eeea.6e0a4] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:50:31 - [info] [OpcUa-Browser:6363eeea.6e0a4] Browse loading Items done ...
1
18 May 17:50:36 - [error] [OpcUa-Browser:6363eeea.6e0a4] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:50:36 - [info] [OpcUa-Browser:6363eeea.6e0a4] Browse loading Items done ...
18 May 17:50:45 - [warn] [OpcUa-Browser:6363eeea.6e0a4] input browser
18 May 17:50:45 - [info] [OpcUa-Browser:6363eeea.6e0a4] start browse client on opc.tcp://127.0.0.1:49320
21:50:45.541Z :message_builder_base          :153   Error   client71 cannot construct object with nodeID ns=0;i=449
1
1
18 May 17:51:19 - [error] [OpcUa-Browser:6363eeea.6e0a4] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
18 May 17:51:19 - [info] [OpcUa-Browser:6363eeea.6e0a4] Browse loading Items done ...

mikakaraila, 2020-05-19:
Fix this first, add client to trusted
18 May 17:49:42 - [error] [OpcUa-Client:82f9e500.e28b98] OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.

youngkzy, 2020-05-19:
these nodes and the OPC Server is configured to have NO SECURITY.  That is why it is working in the aforementioned nodes...  But I will see that they are in the trusted store... Doing that now.

youngkzy, 2020-05-19:
OK, Kepware OPC UA Trusted Servers has Server Cert and Client Cert in 

19 May 13:13:10 - [info] [OpcUa-Client:32a82e31.12b312]         No certificate used.
19 May 13:13:10 - [info] Started flows
19 May 13:13:10 - [info] [tcp in:78cdb0e7.d8d6a8] listening on port 1026
19 May 13:13:10 - Connection successful for database EDB_Foxconn_eCMMS with user Sa
17:13:11.018Z :message_builder_base          :153   Error   client2 cannot construct object with nodeID ns=0;i=449
19 May 13:14:10 - [error] [OpcUa-Client:32a82e31.12b312]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))

youngkzy, 2020-05-19:
Hello, Making sure Certs are OK in Kepware and in Windows Cert manager.  Both KEPServerEX/UA Client Driver and KEPServerEX/UA Server in KEPServer UA Manager are OK.  They are both listed in the Trusted Root Certification Authorities and are "OK".  I rebooted my windows 10 PC.  The same error comes up. 
---------------------------------------------------------------------------------------------

19 May 13:21:19 - [info] [OpcUa-Client:32a82e31.12b312]         No certificate used.
19 May 13:21:19 - [info] Started flows
19 May 13:21:19 - [info] [tcp in:78cdb0e7.d8d6a8] listening on port 1026
19 May 13:21:19 - Connection successful for database EDB_Foxconn_eCMMS with user Sa
17:21:19.989Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449
19 May 13:22:19 - [error] [OpcUa-Client:32a82e31.12b312]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))

mikakaraila, 2020-05-19:
It is inside Kepserver settings

youngkzy, 2020-05-19:
is what inside Kepserver settings?  I checked the certificate store.. that's what the first paragraph is saying Microsoft and Kepserver UA settings use the same dialog for the certs

youngkzy, 2020-05-19:
I validated the tag with a OPC UA channel type that points to an internal DA simulator tag.  and it is correctly monitored in the Quick Client/ OPC Client tester.  I can read these tages with  the other node and other opc test clients.  they are all LOCAL HOST

youngkzy, 2020-05-19:
trying to uninstall nodes and it's giving me errors.  
-----------------------------------------------------------
2020-05-19T20:08:53.514Z Remove : node-red-contrib-opcua

2020-05-19T20:08:53.529Z npm.cmd remove --no-audit --no-update-notifier --save node-red-contrib-opcua

mikakaraila, 2020-05-20:
RTFM:
[image]

youngkzy, 2020-05-20:
Thanks for all your help, but this is not the problem.  At first it was, but the other node did not need this in order to work anonymously.  By the way what document are you referencing?  Please.

Thank you so much for your assistance!

[image]

[image]

coecluj, 2020-05-20:
I have the exact same issue since I upgraded to the latest version. Nothing in my Kepware settings was changed, my client is trusted, but I keep getting the same errors. My application worked for over 3 months, and suddenly it doesn't connect anymore.

youngkzy, 2020-05-20:
@coecluj, 
 Thank you for your input. Perhaps we can figure out what version to go back to...
 I have an application we are working on that I need to install an OPCUA node.  The one I have been using that still works is not available for download and I'm thinking this is the one I need to get working.  I have a Windows10 system, by the way.

@mikakaraila,  What error is this? 

12:07:13.178Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449

coecluj, 2020-05-20:
So I downgraded to version 0.2.53, which had the same issues. Then, I looked into the commits history and found one that [Updated node-opcua to 2.4.4](https://github.com/mikakaraila/node-red-contrib-opcua/commit/81e975d16ebe1558206cf64ca51a9c14f23eef39). I have downgraded then prior to that update, version 0.2.51, and it's working! I think the thing that got to the issue was the upgrade.

coecluj, 2020-05-20:
In case you need help with downgrading, close node-red from the cmd. Then, go to your local .node-red folder (which should be in C:\Users\<User>\.node-red), and from cmd do the following:
`npm uninstall node-red-contrib-opcua`
Then:
`npm install node-red-contrib-opcua@0.2.51`
Let me know if it works for you. Of course, certain features that were implemented after are not available, but for me it still works as intended.
>

youngkzy, 2020-05-20:
@coecluj, 
 Hey thanks, That's the next step for me.  At least we both can get some traction on fixing this problem. 

@mikakaraila, 
What information do you need from me or anyone else, in order to have us help you identify this issue with Kepware?

Thank you.

youngkzy, 2020-05-20:
@mikakaraila,
 I read where there is a similar issue with Security set to "None" with "node-opcua".  Is it possible to coordinate this as the issue with "node-red-contrib-opcua"?  I will try to upgrade to the newer version again and test with security set at something besides "None" at a later date.  Perhaps I was caught in this with the assumption that security set to "None" was an easy means to just test the software function.

mikakaraila, 2020-05-24:
I wa using this KepServer guide:
https://www.kepware.com/getattachment/ccefc1a5-9b13-41e6-99d9-2b00cc85373e/opc-ua-client-server-easy-guide.pdf

And you have accepted None on KepServer?
[image]

mikakaraila, 2020-05-24:
Second item to check! You should see node-red OPC UA Client on this list:
[image]

youngkzy, 2020-05-24:
@mikakaraila,
 Thank you for your response. 
  Yes, I actually have a previous version working ... 2.51, with no security.  Please acknowledge that I have sent this information in prior posts. It seems that since upgrading "node-opcua" 2.44 has perhaps caused an issue.  I believe that all the clients and servers are in each others trust in Kepware and Windows Stores. Since the downgrade, I have seen in Kepware Server side presence of a client reference to node-opcuaTEST or something.  I have trusted that as well.  I'm not having too much success with the security above "none" at present.  Installed and tested with UAExpert and have communicated with the UA Server successfully with signed security,

mikakaraila, 2020-05-25:
Test latest from npm v0.2.61 it contains additional securityPolicies and it will use client certificate.
I tested those with Prosys Simulation server and now client certificate was visible first as untrusted.

youngkzy, 2020-05-26:
I uninstalled V 2.51 and Installed V 2.61 and I have seen good results for level "None" security.   I will test further using security other than "None" later.

 Thank you for addressing this issue.

youngkzy, 2020-07-29:
@mikakaraila,
    So I had 2 versions installed at the same time.  I uninstalled all "node-red-contrib-opcua" and now I have the latest version as of today.  I tested it with the local Kepware.  I can open up a local test client and it see's the server and data.

Here is my results from the new install and test run

[{"id":"f6f3e5da.4b9058","type":"inject","z":"913963d0.5ef3a","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":190,"y":820,"wires":[["8649d164.6690c"]]},{"id":"8649d164.6690c","type":"OpcUa-Item","z":"913963d0.5ef3a","item":"ns=2;s=Simulation Examples.Functions.Ramp1","datatype":"Int32","value":"0","name":"READ","x":330,"y":820,"wires":[["1c6efc24.cf548c"]]},{"id":"1c6efc24.cf548c","type":"OpcUa-Client","z":"913963d0.5ef3a","endpoint":"940c9eb4.da3ac","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","securitymode":"None","securitypolicy":"None","name":"READ ENDPOINT","x":530,"y":820,"wires":[["234152b5.2f6896"]]},{"id":"234152b5.2f6896","type":"debug","z":"913963d0.5ef3a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":820,"y":820,"wires":[]},{"id":"940c9eb4.da3ac","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://127.0.0.1:49320","secpol":"None","secmode":"None","login":false}]

[image]

I'm going back to 0.2.61 for now.  That's all that seems to work.

Kind Regards,
@youngkzy

mikakaraila, 2020-07-29:
Ok I can compare those versions. But node-opcua has also updated. I can't remember when there was update from JavaScript to Typescript.

youngkzy, 2020-07-29:
I actually had to go back to 0.2.51 to get it to work again.   I can probably migrate to 0.2.61 and have it still work as last time.

mikakaraila, 2020-07-30:
It is based on very old JavaScript based node-opcua 0.7.4 and latest one is 2.8.0 so there is something broken on change to typescript based node-opcua API.

mikakaraila, 2020-08-03:
We have duplicates for this one. I will close this one.

closed 2020-08-03 by mikakaraila

ashok0617, 2021-10-07:
@mikakaraila , I was facing this issues with KEPServerEX v6.10 with default project. I could not understand this as similar settings was working on v5.21. I had installed v6.10 on my Windows 10 Home. Thought it will helpful for people facing same issue.

Error: Cannot find ANONYMOUS user token policy in end point description
Reason: By default Kepware project has Allow Anonymous User set No
Fix: 
- Go to System tray->KEPServerEX (icon)->Right Click->Configuration
- On configuration app open project properties by selecting the project->right click->properties
- Locate Allow Anonymous User and set Yes (By default, it is No)
[image]

Note: This issue not related to certificate at all as security policy set to null for server endpoint
[image]

For me client was added as trusted by default 
[image]

Security policy for anonymous client was default(allowed)
[image]

I could not find until look for respective v6.10 Kepware doc and join the dots
https://www.kepware.com/getattachment/e1943820-ef3c-4932-b055-4ef2a80ab863/kepserverex-manual.pdf
Search: Anonymous login
Page 40 - Client Sessions