mikakaraila/node-red-contrib-opcua #152: server options are not defined

pfetrial, 2020-05-20:
hello 
I am trying to use the server and the sample flow using the server but If I  deploy the flow the server says "not running" and in the log I see : 20 May 08:57:31 - [error] [OpcUa-Server:c27855fe.a75c78] ReferenceError: server_options is not defined.
How to run the server? thanks for your reply.

mikakaraila, 2020-05-20:
Already fixed to v0.2.58

closed 2020-05-20 by mikakaraila

pfetrial, 2020-05-20:
ok thanks. I've updated the nodes. now I have 👍 20 May 14:53:47 - [red] Uncaught Exception:
20 May 14:53:47 - TypeError: Cannot read property 'endpointDescriptions' of undefined
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua\104-opcuaserver.js:304:41
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:2955:19
    at wrapper (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:268:20)
    at iterateeCallback (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:413:21)
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:321:20
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:2953:17
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\opcua_server.ts:1181:13
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\opcua_server.ts:1255:11
    at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\base_server.ts:219:17
    at wrapper (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:268:20)

sorry..