mikakaraila/node-red-contrib-opcua #153: Cannot read property 'endpointDescriptions' of undefined

pfetrial, 2020-05-23:
hello. I've updated the all the nodes on latestest version. now I have :
20 May 14:53:47 - [red] Uncaught Exception:
20 May 14:53:47 - TypeError: Cannot read property 'endpointDescriptions' of undefined
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua\104-opcuaserver.js:304:41
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:2955:19
at wrapper (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:268:20)
at iterateeCallback (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:413:21)
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:321:20
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:2953:17
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\opcua_server.ts:1181:13
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\opcua_server.ts:1255:11
at C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-server\source\base_server.ts:219:17
at wrapper (C:\Users\Patrick\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\async\dist\async.js:268:20)

What is the problem now ?
Thx

mikakaraila, 2020-05-23:
Still missing parameter, I will look it tomorrow

mikakaraila, 2020-05-24:
It was using old API, updated code. New version is 0.2.59

closed 2020-05-24 by mikakaraila