mikakaraila/node-red-contrib-opcua #154: Unsubscribe don't work

gogolevsergey, 2020-05-25:
I have this flow, but unsubscribe method doesn't work.
[image]

`
Got:undefined
25 May 09:15:34 - [error] [OpcUa-Client:4fa20999.f0b528] Item not monitored:ns=5;s=MyLevel
`
I found similar bug #106  fixed year ago, it's a regression? 

Also found workaround - send action:"reconnect"
[image]

but this workaround have more bigger problem it don't release OPC UA Session after reconnect. So after each call sessions stay alive in server, and acknowledges as alive.

[image]

this can lead to server problems (for example after 50 sessions i got " OPC UA connection error: Too Many Sessions : BadTooManySessions (0x80560000)")

mikakaraila, 2020-05-25:
Most probably update to typescript based node-opcua has changed API. I will look later today.
Session re-connect is known to be problematic.
Which version you are using?

gogolevsergey, 2020-05-25:
latest 

"node-red": "1.0.6"
"node-red-contrib-opcua": "==0.2.59"

mikakaraila, 2020-05-25:
Done, need to fix some other parts as well.
Now unsubscribe must be done to same node as has done subscription with inject:
[image]

mikakaraila, 2020-05-25:
I added into the code session close. It should prevent active sessions overflow.
Some servers limit maxSessionNumber it could cause situation you have faced.
Please test. There is newer version on npm. Latest one is not yet on node-red flows.

gogolevsergey, 2020-05-26:
Now status: "invalid endpoint opc.tcp://192.168.31.121:53530/OPCUA/SimulationServer"
and no connection

logs:
`
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] Client status: invalid endpoint opc.tcp://192.168.31.121:53530/OPCUA/SimulationServer,
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] No client to connect...,
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] Client status: connecting,
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] async series - connecting ,
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] Client status: create client,
26 May 18:51:24 - [error] [OpcUa-Client:2bd282ae.80167e] Cannot create client, check connection options: {"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#None","securityMode":1,"certificateFile":"/usr/src/node-red/node_modules/node-opcua-client/certificates/client_selfSigned_cert_2048.pem","privateKeyFile":"/usr/src/node-red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","endpoint_must_exist":false,"defaultSecureTokenLifetime":40000,"connectionStrategy":{"maxRetry":10512000,"initialDelay":2000,"maxDelay":30000},"keepSessionAlive":true},
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] [object Object],
26 May 18:51:24 - [warn] [OpcUa-Client:2bd282ae.80167e] OpcUaClientNode: create Client ...,
26 May 18:51:24 - [info] [OpcUa-Client:2bd282ae.80167e] Using client certificate /usr/src/node-red/node_modules/node-opcua-client/certificates/client_selfSigned_cert_2048.pem
`

mikakaraila, 2020-05-26:
Are you using None None security mode/policy? What is on simulation server?

I tested it with Prosys UA Simulator.
Something is different...

mikakaraila, 2020-05-28:
Did you merge changes?
[image]

gogolevsergey, 2020-05-30:
> Are you using None None security mode/policy? What is on simulation server?
> 
> I tested it with Prosys UA Simulator.
> Something is different... I used Windows you're using Linux.

yes on windows it's working, but on linux it won't work at all now, but before it's worked.

OPC UA Server - Prosys Simaulation Server, security policy/mode - None/None.

[image]

mikakaraila, 2020-05-31:
And you can access endpoint above with UaExpert?

gogolevsergey, 2020-06-01:
yes, UaExpert connect succesfully. Files present, I start node-red in docker-compose with these files: 
[nodered-opcua-collector.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/4711162/nodered-opcua-collector.zip)

gogolevsergey, 2020-06-02:
Just tested with version 0.2.62 all work OK as expected, thanks for help

closed 2020-06-02 by gogolevsergey