mikakaraila/node-red-contrib-opcua #155: Number of Tags limitations the node OPC UA can subscribe

henjoe, 2020-05-28:
Hello, I just want to ask if there is a number of limitations of tags to read does subscribe method ? 
For example in my OPC UA Server I have a thousands of Tags, and I want to read all those tags using the UA-Client node (each tags has 1 UA client with a subscribe as method). Will it be a good practice to have 1000 OPCUA Client node for this? or it has a limitations? 

Thank you.

mikakaraila, 2020-05-28:
Client is not limiting, server can limit. I have tested several thousands of items.
TIP: Use one client node, read config from file and use function node to format msg to configure all needed items. There is also multiRead & multiWrite, but subscriptions is most probable giving best performance.

henjoe, 2020-05-29:
Thanks for prompt answer. Never known about the confif file that the UA client can read, I mean that when I installed the OPC UA node, there is no config file or config node. May i know where I can find it? Or maybe any instruction how to do it. 

Again, thanks mate.

mikakaraila, 2020-05-29:
Use standard file node read line by line and inject msg to client node. Very simple. I can do example during weekend, but try yourself.

henjoe, 2020-05-29:
Oh see. That's nice then. 
Is there any syntax does the config file to be? 
Or I just put all my Tagnames line by line there?

Thanks Sir.

mikakaraila, 2020-05-31:
Copy prosys.txt into .node-red folder and flow file also.
[prosys.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4708084/prosys.txt)

Rename this NODEIS_FROM_FILE.txt to .json
[NODEIDS_FROM_FILE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4708085/NODEIDS_FROM_FILE.txt)

henjoe, 2020-06-01:
Thanks Sir, 
Will try this one. 

With kind regards,
Henjoe

henjoe, 2020-06-17:
Hello Sir, I know this was already close, but I just wanna ask if what could be the reason on this. 
Since I did the 1 OPC client node and subscribe to 1000 tags in 1 secs interval, all my Tags are updating every seconds and this works fine. However, I noticed that some Tags even though they update every seconds, the output of that Client node skips a seconds, it is like:
12:01:00 PM 25.23
12:01:01 PM  28.24
12:01:04 PM 23.74

There is a 2 seconds skip on data. And this continue at random time.
What I did is I just try to subscribe obly for 1 Tag, the same tag I saw that was skipping, and the output of UA client bode doesn't skip anymore, it updates every seconds. 

I just dont know if this is related to my configuration on node, or just the UA node cannot handle this amount of data with per second update?

Again, thank you very much. 
I highly appreciate all your works. 

God bless and Stay safe.

Regards

mikakaraila, 2020-06-18:
This depends on server implementation also. Please test server performance with UaExpert. Another option is to use read multiple and inject interval

henjoe, 2020-06-18:
Hello Sir, 
I am confident that the server works fine, already checked it on opcuaExpert, all data are updating per second.

I will try later the read multiple data, but I think this kind of reading doesn't have sourcetimestamp value right? Source timestamp is somehow important to me. 

Again, thank you sir.

mikakaraila, 2020-06-18:
There was some issue concerning timestamps on node-opcua some time ago...

Another comment are you sure you have made subscription with 1s interval?
Use node-red -v then you can check it from debug panel.

henjoe, 2020-06-18:
Hellk Sir,

What I did is since i have 1000 Tags, i used to distribute it in 5 uanode clients, basically there are 200 tags per ua client. I already used the node-red -v and see thr debug windows. I noticed that there is a message saying " Tagname succesfully subscribed with 100ms interval". I am sure that the settings on interval was set to 1000ms. 

I just checked it again, I am actually confused right now, it is now okay. The timestamp is giving me every secs.
I noticed that when i run nodered with this command "nodered -v", the problem of timestamp occur, but when I used command "node-red" only, the output is okay. 
It's kinda weird, does -v parameter has something to so with it?

closed 2021-06-10 by mikakaraila

Theodorska1, 2023-08-14:
Hello, 

I have an array of 150 node-ids. When i push them all to the client node in subscribe it only subscrives to 7 of them, Especially those that are folders. When i make the list array f.x. 14 (with the same data) it has no problem subscribing to all indivdual tags and folders.

mikakaraila, 2023-08-14:
Check this example: it will first build server with variables then client will use them and 
make performance measurement: 1000 msg/sec.

Flow file:
[OPCUA-CLIENT-PERFORMANCE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12334614/OPCUA-CLIENT-PERFORMANCE.txt)

Theodorska1, 2023-08-17:
> Check this example: it will first build server with variables then client will use them and make performance measurement: 1000 msg/sec.
> 
> Flow file: [OPCUA-CLIENT-PERFORMANCE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12334614/OPCUA-CLIENT-PERFORMANCE.txt)
> 
> Just extra note: With node-opcua I have build aggregator with 100 000 variables and client can read all them in 1 sec. Code is not node-red flow, own compiled nodejs application.

Thank you so much for this. 
I now have the correct array of 412 nodeids, but when i try to subscribe to them only between 5 to 20 gets subscribed to. They are always the same values and from different locations in the array. I have tried subscribing to some of the other nodeids in isolation from the ones that show up and they work. I even tried adding a delay to add one by one to the opc client node. However, i still get the same ones when i am attempting to subscribe to all. Any idea why this might be?

Theodorska1, 2023-08-17:
I solved my issue, when extracting the node IDs the last part of most of the node paths did not contain a capital letter at initial. 
f.x. ns=4;s=PMS.Test.test had to be converted to ns=4;s=PMS.Test.Test

Theodorska1, 2023-08-21:
We still have some issues that certain letters in the nodeID that are capitalized in our OPC server does not have their capitalization when read from the  OPC UA Browser node. Therefore I have to manually figure out where I loose capitalization and write a script to change the capitalisation of certain nodeID's to enable the OPC UA Client node to subscribe to that nodeID. Is this a bug or am I doing something wrong?

f.x. "...CB_UPS_Bat_Tripped" comes out of the browser node as "...CB_UPS_bat_tripped", which the client does not recognize as a valid nodeID.

Theodorska1, 2023-08-21:
It is some case-sensitivity issue when a browsed nodeID contains multiple sub-values, like a folder