mikakaraila/node-red-contrib-opcua #158: OpcUa-Client TypeError [ERR_INVALID_ARG_TYPE] (Version 0.2.62)

stfriess, 2020-06-04:
I use the OpcUa-Client node to connect to our OPC-UA server.
Right after deploying my flow I get this error message:

4 Jun 10:44:19 - [error] [OpcUa-Client:2ed4e3e.b63391c] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type number (-1)

This is the configuration of my OPC-UA client node:
[image]

Countercheck with UAExpert: connection to our OPC-UA server works fine:
[image]

An older version worked fine. Guess it was 0.2.53 or so.
Any idea what I could do?
Thanks for your help, Stefan

stfriess, 2020-06-04:
This is my node-red/Node.js confiugration:
[image]

mikakaraila, 2020-06-04:
Ok most probably still something after update to Typescript, I will try to find some time to look this tomorrow or weekend

mikakaraila, 2020-06-05:
Can you provide more information? 
Use command: node-red -v TEST.json
There is one mismatch as you have security: None/None and 
still using credentials with username&password.

mikakaraila, 2020-06-05:
My environment is working with similar config:
[image]

stfriess, 2020-06-05:
[image]

I created a similar flow like yours

stfriess, 2020-06-05:
[image]

stfriess, 2020-06-05:
OPC UA user and password is same on my system: "test"

mikakaraila, 2020-06-05:
Can you check where is client certificate file?
client_selfsigned_cert_2048.pem

stfriess, 2020-06-05:
I have 3 of them. 
1. C:\Users\sriess\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\certificates
2. C:\Users\sriess\.node-red\node_modules\node-opcua-client-proxy\node_modules\node-opcua-client\certificates
C:\Users\sriess\node_modules\node-opcua-client\certificates
Do I have misconfigured something? Thank you for your help

mikakaraila, 2020-06-06:
Only first should be used, can you rename 2&3 folder name to test if it helps?

stfriess, 2020-06-06:
I renamed the folders 2 and 3 to ~certificates. Unfortunately same error message.
Is the folder structure ok from your point of view? I think about re-installing node-red and packages from scratch.

mikakaraila, 2020-06-06:
Is there \ before .node-red on first?

stfriess, 2020-06-08:
Sorry for the delay. Yes, there is a '\' in the path in friont of .node-red:
C:\Users\sriess\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\certificates

stfriess, 2020-06-08:
[image]

stfriess, 2020-06-08:
If I set clientPkg to a fixed path, it is running:
    var clientPkg = "C:/Users/sriess/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client";

[image]

But afterwards, I get this message:
[image]

Still wonder, why getInstalledPathSync() does not lead to the correct path. Maybe really a configuration/installation problem

mikakaraila, 2020-06-09:
Or it returns an array of folders as you have multiple of those installed:
Code tries to find installedPath.getInstalledPathSync('**node-opcua-client**',

I suggest you to remove not needed parallel installations.

stfriess, 2020-06-09:
Ok, I do. One last question: is there any difference between installing
packages directly via node-red and per command line by npm?
Thank you very much for your help!
Best regards, Stefan

mikakaraila, 2020-06-12:
node-red manage palette install them to correct place.

closed 2020-06-17 by mikakaraila