mikakaraila/node-red-contrib-opcua #160: "addVariable" injected to OPC UA Server node outputs error

hilschernetpi, 2020-06-11:
I have this very simple Node-RED flow injecting an addVariable opc ua command to the server node.

[{"id":"387c03f0.5e16cc","type":"debug","z":"96af5d97.591158","name":"","active":true,"console":"false","complete":"false","x":810,"y":420,"wires":[]},{"id":"281047e4.431938","type":"OpcUa-Server","z":"96af5d97.591158","port":"53880","name":"","x":563.5,"y":422,"wires":[["387c03f0.5e16cc"]]},{"id":"3c25258b.257dba","type":"inject","z":"96af5d97.591158","name":"New variable (Int32)","topic":"ns=4;s=TestInt32;datatype=Int32","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":290,"y":420,"wires":[["281047e4.431938"]]}]

But instead of creating a new variable it outputs an error in the debug window of Node-RED.

I am using Node-RED V1.0.6 and the latest opc ua node in the version 0.2.63 on a Linux machine.

Is there anything I need to inject prior adding a variable that it runs or is is a bug?

mikakaraila, 2020-06-12:
Use namespace index 1. I expect node-opcua is more strict with address space. If namespace is not defined it cannot be used.

[image]

hilschernetpi, 2020-06-12:
Oh yes it works perfect now with n=1

Just to highlight why I was using n=4 first of all. I was using your example from your github repo https://raw.githubusercontent.com/mikakaraila/node-red-contrib-opcua/master/examples/OPCUA-TEST.json.  It should be updated to use n=1.

closed 2020-06-12 by hilschernetpi