mikakaraila/node-red-contrib-opcua #161: OPC UA server improvement: injecting msg when data was set from OPC UA client

hilschernetpi, 2020-06-15:
Today the OPC UA server does not send an indication msg when an OPC UA client has written a value to an added variable.

This can be improved adding two more lines in the function set() in the 104-opcuaserver.js file

 ```
set: function (variant) {

  variables[browseName] = opcuaBasics.build_new_value_by_datatype(variant.dataType, variant.value);

  var SetMsg = { "payload" : { "messageType" : "Variable", "variableName": browseName, "variableValue": variables[browseName] }};
  node.send(SetMsg);

  return opcua.StatusCodes.Good;
}
```
So next to writing the value to the variables[ ] as before it is now also sending an indication message to the OPC UA server user who can then grab the data value for own purposes. The msg has the same format as the message used to set the value of a variable.

Maybe it is worth it to embed it in the next version

mikakaraila, 2020-06-15:
Thanks, I will add this.

mikakaraila, 2020-06-22:
Published new version

closed 2020-06-22 by mikakaraila