mikakaraila/node-red-contrib-opcua #162: Issue Bad to Many Sessions (0xe8056000)

leonvoss11, 2020-06-17:
Hello,

I have 40 nodes that I want to read and 10 nodes I want to write. For this I used the client function with read and write. For writing I used one client block for writing and one client block for reading per read access.  Now I get an error message Bad to Many Sessions (0xe8056000) I think this is due to the many read accesses. 
Would the solution be read mutiple ? how can I select the values again afterward so that I can work with them?

mikakaraila, 2020-06-17:
If I remember correctly it is already implemented. I will look example...

leonvoss11, 2020-06-17:
ok Tanks

mikakaraila, 2020-06-22:
Here is example how to use file to define nodeIds with datatypes and read them with one action.

[NODEID_json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815025/NODEID_json.txt)

NOTE: Last line is empty, needed to run action as last msg.
[nodeids.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815027/nodeids.txt)

mikakaraila, 2020-06-22:
Here is extended version:
[image]

[NODEID_json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815068/NODEID_json.txt)

[nodeids.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815071/nodeids.txt)

mikakaraila, 2020-07-07:
And server can limit sessions to 10 or even lower amount

closed 2020-07-07 by mikakaraila