mikakaraila/node-red-contrib-opcua #163: Termux

korpmaks, 2020-06-17:
Try it on Android:
"Error:  cannot locate certificate file /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem"

Made this:
openssl genrsa -out /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem -des3 -rand /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/config.js 2048
"Error:  cannot locate private key file /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem"

Made this:
openssl genrsa -out /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem -des3 -rand /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/certificates/config.js 2048

Then node-red interupted:
[red] Uncaught Exception:
18 Jun 00:01:11 - Error: Invalid position : buf.length=0 pos =0
    at Object.readTag (/data/data/com.termux/files/home/.node-red/node_modules/node-opcua-crypto/dist/asn1.js:34:15)
    at Object.split_der (/data/data/com.termux/files/home/.node-red/node_modules/node-opcua-crypto/dist/crypto_explore_certificate.js:514:35)
    at OPCUAClientImpl.getCertificate (/data/data/com.termux/files/home/.node-red/node_modules/node-opcua-common/dist/opcua_secure_object.js:37:52)
    at OPCUAClientImpl._getApplicationUri (/data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/dist/private/opcua_client_impl.js:743:34)
    at OPCUAClientImpl.__createSession_step3 (/data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/dist/private/opcua_client_impl.js:555:37)
    at /data/data/com.termux/files/home/.node-red/node_modules/node-opcua-client/dist/private/opcua_client_impl.js:539:18
    at cb (util.js:207:31)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)

mikakaraila, 2020-06-18:
I suppose this is more related to node-opcua at Android

mikakaraila, 2020-07-07:
Look my latest changes related to getInstalledPath this needs additional "local" path to look for...

erossignon, 2020-11-04:
you will need to install openssl package on termux. openssl is not installed by default on termux.