mikakaraila/node-red-contrib-opcua #164: node-red-contrib-opcua Subscribing and EcoStruxture OPCUA Server Expert

Unipir81-OG, 2020-06-18:
Firstly thank you for creating these nodes. These have a huge potential!!

Unfortunately I have come across a little snag. When using the Prosys server it would seem on the surface everything works.

The problem arises when I switch to another OPCUA server, in this case EcoStruxture OPCUA Server Expert from Schneider-Electric.

In the most part reading individual items from this server works perfectly. The problem is when attempting to utilize Subscribe and Monitor with the OpcuaClient node. I have attached the flow I am doing this basic testing with as well.

The attached are the errors returned in the Node-RED server when attempting to use these functions of the OpcuaClient. Once they occur the entire Node-RED server fails.
Am I implementing something incorrectly when using these nodes? Or is this a possible bug in the node themselves?

Your help is greatly appreciated! If you require any further input from me please let me know in reply.

Rob

P.S. not going to lie our team is new to both Node-RED and OPCUA so please be gentle
[Node-RED_OpcUAClient_OPCUA-ServerExpert.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4796825/Node-RED_OpcUAClient_OPCUA-ServerExpert.txt)
[Node-RED_OpcUAClient_Error.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4796745/Node-RED_OpcUAClient_Error.txt)

mikakaraila, 2020-06-18:
Are you using v0.2.63 or older?

I made small test flow:
[image]

[TEST2_json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4797491/TEST2_json.txt)

Unipir81-OG, 2020-06-19:
Mika

Thank you so much for the quick response! I will work on testing the newer version in the coming days.

I will keep you posted on results.

Cheers

Robert Nabholz
Control & Monitoring Coordinator | AUSTAL AUSTRALIA
100 Clarence Beach Road Henderson Western Australia 6166
T: +61 8 9410 1111 | M: +61 (0)427 713650 | D: +61 8 6414 6957 | E:  robert.nabholz@austal.com<mailto:robert.nabholz@austal.com>
www.austal.com<http://www.austal.com/>

SHIPS.SYSTEMS.SUPPORT

[image]<https://user-images.githubusercontent.com/5184524/85000065-97ca1e00-b15a-11ea-9866-0cad887d29a8.png>

The information contained in this email is confidential to the Austal Group of Companies and
may also be subject of legal professional privilege. If you are not the intended recipient, or
responsible for delivery of the message to such a person, any use, disclosure or copying of
this email is unauthorised. If you have received this email in error, please notify the sender
by return email and delete the message. Please advise the Austal Group IT Department
(servicedesk@austal.com) immediately if you or your employer do not consent to Internet email for
messages of this kind. All reasonable steps have been implemented to ensure the accuracy and
integrity of all electronic communications, but no liability is accepted for any materials
transmitted.

mikakaraila, 2020-06-23:
I would change inject date to inject string (empty). Code uses date payload as parameter (huge number as a subscription interval).

Unipir81-OG, 2020-06-24:
Mika

Apologies for not responding earlier! Updating the version most definitely corrected the issue of concern.

I may have a few more queries in regard to the data returned from the OPC UA Server but I will begin a new thread for this on Git.

Cheers

Robert Nabholz
Control & Monitoring Coordinator | AUSTAL AUSTRALIA
100 Clarence Beach Road Henderson Western Australia 6166
T: +61 8 9410 1111 | M: +61 (0)427 713650 | D: +61 8 6414 6957 | E:  robert.nabholz@austal.com<mailto:robert.nabholz@austal.com>
www.austal.com<http://www.austal.com/>

SHIPS.SYSTEMS.SUPPORT

closed 2021-06-10 by mikakaraila