mikakaraila/node-red-contrib-opcua #165: TypeError: ClientSession is not a constructor

jzhvymetal, 2020-06-18:
OpcUA-Client Node always give TypeError: ClientSession is not a constructor.  My flow was adapted from the example and attached below.  Read and Write works without a problem.  

I'm trying to create a function with info to to get proper data type to read all values within object.  It would be nice if the browse would also include data types.

`[{"id":"d4ab38ed.0acd48","type":"OpcUa-Client","z":"a6dc184b.242038","endpoint":"afd6c31e.26bde","action":"info","deadbandvalue":"","time":10,"timeUnit":"s","localfile":"","name":"ProSys SimServer","x":790,"y":640,"wires":[["41261ef6.58e1f"]]},{"id":"ef403fde.8dc0d","type":"OpcUa-Item","z":"a6dc184b.242038","item":"ns=2;s=GVL_OPC.myReal","datatype":"Float","value":"","name":"Static Double","x":582,"y":642.4000091552734,"wires":[["d4ab38ed.0acd48","623ad172.ec495"]]},{"id":"41261ef6.58e1f","type":"debug","z":"a6dc184b.242038","name":"","active":true,"console":"false","complete":"false","x":1019.9999847412109,"y":638.8000030517578,"wires":[]},{"id":"17b13374.96d9bd","type":"inject","z":"a6dc184b.242038","name":"Test Double","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":"","x":383.99998474121094,"y":642.6000061035156,"wires":[["ef403fde.8dc0d"]]},{"id":"b7da53c2.e2cfc","type":"comment","z":"a6dc184b.242038","name":"INFO","info":"","x":200.5,"y":644.4000091552734,"wires":[]},{"id":"623ad172.ec495","type":"debug","z":"a6dc184b.242038","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":760,"y":560,"wires":[]},{"id":"afd6c31e.26bde","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://192.168.10.241:4840/","secpol":"None","secmode":"NONE","login":false}]`

mikakaraila, 2020-06-19:
Ok have to check what node-opcua provides...
Are you using v0.2.63?

jzhvymetal, 2020-06-19:
Yes running the latest v0.2.63.  Previous version worked but never downgraded to verify it still worked.

Thanks.

mikakaraila, 2020-06-22:
I expect this one was related to browse node? I made changes as pointed at #166

mikakaraila, 2020-06-23:
I expect new browse v0.2.64 fixed this one.

closed 2020-06-23 by mikakaraila